RobotLocomotion/drake #8489: questions about the use of accuracy in Context

RussTedrake, 2018-03-31:
I've been digging in a bit to the use of accuracy in the Context, and have some questions (concerns?).  
  1) it seems appropriate to set accuracy at the LeafSystem/Subsystem level -- i might understand the accuracy required for accurately simulating my e.g. RimlessWheel, but the only accuracy that is used by the Simulator is at the System level.  There do not appear to be any mechanisms that keep these in sync.  
  2) related, if a LeafSystem wants to use the currently set accuracy in it's methods (as proposed by @edrumwri in my rimless wheel PR), then it cannot do this if the only accuracy that was set was at the Diagram level.  
  3) the documentation (http://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_context.html#a0512ea9a61f4373793f55fb7594dfd1b) indicates that the accuracy can and should be used in multiple ways by multiple methods.  But that seems like a defect to me, because the units of the accuracy may be different in different cases.  For instance, until I dug into the code, I did not expect the accuracy used by the witness isolation to have units of time.  How can that be the same number that we use for e.g. variable step integration?  
 4) despite that documentation, it looks like the only use (that at least CLion "find usages" could uncover) is by the witness isolation?  are there more in the works?  or am I missing some?  Do we not actually have any error-controlled integrators available yet?

edrumwri, 2018-03-31:
Answers to 1 and 2: the plumbing from the diagram context to the leaf system context has needed to be hooked up. I've done this in my rod2d development branch, where I've encountered many such issues before.

3. The API has not been fully fleshed out here, as you see, but we had a few use cases that we wanted to support:

* User only sets the accuracy in the context, which would apply to everything: integrators, systems, trajectory optimization, etc.

* User only sets the accuracy in an algorithm (like an integrator).

* Users sets accuracy in both the context and the algorithm, in which case the latter would override the former.

Getting this logic into place has been one of my OKRs for a while, but it's been continually pushed back to the next quarter :)

My thought has been that the accuracy set in the context has a multiplicative effect, is unit-less, and- if not specified- would be equivalent to 1. That would mean that the accuracy has a relative effect, which is what I was aiming for: you could tighten or loosen the accuracy for all systems and algorithms in a single place. 

4. It's currently only used in the witness function isolation (and in my rod2d branch).

RussTedrake, 2018-03-31:
For the witness functions accuracy, i would have assumed it the accuracy was related to the deviation in the witness function value, not in time.  I was surprised to see it having units of time.  Documentating it would help.

sherm1, 2018-03-31:
The `accuracy` in the context is a unitless value. It is just there to convey the user's accuracy request to any internal algorithms that might care. It is supposed to be propagated to all subcontexts whenever it changes -- that's how it works in the caching branch, and accuracy-dependent computations get invalidated. It should be handled identically to `time`, which is also propagated to all subcontexts.

The Simulator uses `accuracy` for a variety of purposes, including variable-step integration, convergence of the nonlinear system in implicit integration, and strictness of witness function isolation. Later we will also use it for constraint projection tolerance. In each case, the relative accuracy value must be converted to suitable absolute units. That is done by obtaining a "characteristic" length, angle, or time and scaling it by the relative accuracy request.

RussTedrake, 2018-03-31:
i hear you.  but do we agree that the *only* current use has units of time?  (the user has no way to report a characteristic time constant).  if it was, instead, the error in the zero crossing, then the user could set the units by their specification of the witness function.

sherm1, 2018-04-01:
There are other uses for `accuracy` now (used in integrator error control) and planned, but I agree that we _should_ have a tolerance around the zero crossing. How and whether to scale that by the relative accuracy isn't clear to me yet. OTOH, the event isolation window in time clearly should be narrower as the integrator is asked for more accuracy. For the witness value (non-time) axis we could collect an absolute tolerance from the witness function designer, or an accuracy-scaled unit value.

RussTedrake, 2018-04-01:
@sherm - do we have an error controlled integrator?

edrumwri, 2018-04-01:
Yes, we have two: the RK3 and the implicit Euler.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

sherm1, 2018-04-01:
RK3 is the default. If you want a fixed-step integrator you have to ask for it.

RussTedrake, 2018-04-05:
Reminder to me (or evan) - once the accuracy mechanism is fixed for shared contexts, i would like to have the RimlessWheel set a default accuracy in e.g. a `SetDefaultContext`.  Right now, every simulation has to remember to set the accuracy (and probably sets it differently).

edrumwri, 2018-04-05:
Got it.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

edrumwri, 2019-12-02:
Update to this issue responding to @RussTedrake's comment:

> i would like to have the RimlessWheel set a default accuracy in e.g. a `SetDefaultContext`.

Now looking in this: Russ, your request isn't workable in mechanics and concept. With regard to mechanics, there is no way to overload `SetDefaultContext()`. With regard to concept, setting the accuracy of a system in isolation generally isn't helpful- it must be done at the Diagram level.

One option might be to allow a system to specify its minimum acceptable accuracy, but I think the relative nature of "accuracy" and its varying interpretation by the analysis algorithms would put a hard burden on the system designer.

PR #12428 (2019-12-03): Updates the documentation on the use of accuracy in the Context.
Provides the final piece to resolve @RussTedrake's issues in #8489. (Resolves #8489)

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12428)
closed 2019-12-03 by jwnimmer-tri