RobotLocomotion/drake #8495: Support Environment dictionary declarations for vector-valued decision variables

gizatt, 2018-04-02:
I definitely think it would be very convenient to have the ability to evaluate symbolic expressions by specifying whole vectors of variables at once! This would be useful for e.g. [evaluating constraints that are a function of a relatively large robot state](https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/solvers/test/pr2_ik_test.py).

@soonho-tri

soonho-tri, 2018-04-03:
@gizatt , I assume that you want to have both of a symbolic variable and a vector of them as keys in a Python dictionary such as the following?

```python
env = {
    x_vars : np.array([1.0, 2.0, 3.0, 4.0])  # x_vars = [x_1, x_2, x_3, x_4]
    y : 3.0
}
... e.Evaluate(env) ...
```

or do you want a simpler signature such as `e.Evaluate(x_vars, np.array([1.0, 2.0, 3.0, 4.0])`?

gizatt, 2018-04-03:
The former -- a dictionary with mixed key types would be optimal (though I understand if it's really tricky).

soonho-tri, 2018-04-03:
I'll check what I can do. 

BTW, @EricCousineau-TRI is working on #8500 which will make conflicts with this request eventually.

RussTedrake, 2021-04-07:
ftr -- i just ran into this one again... definitely wishing we had support for this.

soonho-tri, 2021-04-07:
FYI, neither Python list nor numpy.ndarray is unhashable. As a result, we can't have them as keys in a Python dictionary. What would be a good signature for the proposed method? Maybe one taking a heterogeneous list of variables and values? For example:

```python
expr.Evaluate([
    [x, y, z], [1.0, 2.0, 3.0],
    w, 4.0
])
```

EricCousineau-TRI, 2021-04-07:
Yup, that or even just an iterable of pairs (which is how `dict` can be constructed from items; [see docs](https://docs.python.org/3/library/stdtypes.html#dict)), so it could be a list, tuples, etc.

The other alternative is to make it "hashable" via something like:
```
import pydrake.symbolic as sym

expr.Evaluate({
   sym.key([x, y, z]): [1.0, 2.0, 3.0],
   sym.key(x_vars): 4.0,
})
```

Then `key(x)` could be implemented simply like `def key(x): return tuple(np.asarray(x).flat)` (if we rely on value for storing shape), or you can pass the shape as part of the key, e.g. `def key(x): x = np.asarray(x); return (x.shape, tuple(x.flat))`.

jwnimmer-tri, 2021-11-18:
For the record, the C++ evaluation environment already accepts a list of pairs, or same-sized matrices of (array[keys], array[values]), which are then matched up pairwise.  This a python-only feature request.

jwnimmer-tri, 2021-11-19:
The link in the OP has bitrotted.  As best I can tell, it was to https://github.com/RobotLocomotion/drake/blob/4dd42dad0c77995ccb124583194d13cbbeea34cb/bindings/pydrake/solvers/test/pr2_ik_test.py, but that code doesn't have any symbolic evaluations.

What we need in order to make progress here is a specific example where the sugar is lacking.

Currently, the user could do this, for example -- using a list of pairs:
```python
vars = [...]
values = np.array([...])
env = dict(zip(vars, values))
expr.Evaluate(env=env)

env2 = dict(list(zip(vars, values)) + [[z, 5.0]])
expr.Evaluate(env=env2)
```

This isn't necessarily very compact, but it is Pythonic and should be clear to many readers.  Some choices of sugar would actually be less clear.

Next time we have need of this sugar, please post a (commit-pinned) hyperlink to the lines of code that are overly awkward, and re-open the issue.  Until then, I'll close it as "needs more info".

closed 2021-11-19 by jwnimmer-tri