RobotLocomotion/drake #8498: Building Drake in a Docker Container "typo"

CatchMeFastFat, 2018-04-03:
at http://drake.mit.edu/docker.html
`cd /drake`
`$ bazel run //drake/examples/contact_model:bowling_ball`
`$ bazel run //drake/examples/kuka_iiwa_arm:kuka_simulation`
`$ bazel run //drake/examples/kuka_iiwa_arm`

Should be `$ bazel run //examples/...`

Also in `drake/setup/docker/Dockerfile.ubuntu16.04.***`
`line 8: bazel build //examples/acrobot:acrobot_run_passive`
should be ` bazel build //examples/acrobot:run_passive`

BTW: Why python in drack visualizer import pydrake while in normal console gives "ImportError: No module named pydrake"

Thanks.

jwnimmer-tri, 2018-04-03:
Assigning to @EricCousineau-TRI as related to #7654.

EricCousineau-TRI, 2018-04-03:
Thanks for letting us know! I've just submitted #8501 to fix this.

> BTW: Why python in drake visualizer import pydrake while in normal console gives "ImportError: No module named pydrake"?

This is because the Bazel version of `drake-visualizer` does not include `pydrake`, as it greatly increases the build time.

Can I ask what your use case for `pydrake` inside of `drake-visualizer` is? Is it something you could use a [packaged binary of Drake](http://drake.mit.edu/from_binary.html) for?

If not, we could provide a `drake_visualizer` flavor which also includes `pydrake`.

CatchMeFastFat, 2018-04-03:
I get this repeated error when testing:
`error: cpp_template_test.py uses drake_py_unittest but is marked executable in the filesystem; fix this via chmod a-x cpp_template_test.py`

I meant that in drake-visualizer' python console when I run pydrake, it gives no error, but in normal bash python it gives "ImportError: No module named pydrake".
Do I have to first open drake-visualizer then run my code? I want a newbie example for using inside docker without network and use MIT 6.832 examples.

Regards

closed 2018-04-03 by sammy-tri

EricCousineau-TRI, 2018-04-03:
Not sure why you're getting those errors, as `cpp_template_test.py` is not marked executable in Git (at least in the version that I have cloned).
Would you be able to make a StackOverflow post about the `cpp_template_test.py` issue, use the `drake` tag, and post what platform you're generating your Docker image from?
(Seems like the file's executable bit is getting set via something like Cygwin in Windows, copied to the Docker image, and then failing the `drake_py_unittest`.)

For using `pydrake` from Python in a normal Bash session, you need to ensure that your `PYTHONPATH` is set correctly:
http://drake.mit.edu/python_bindings.html#using-the-python-bindings

PR #8501 (2018-04-03): doc/docker: Fix typos
Closes #8498

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8501)