RobotLocomotion/drake #8503: Per bazel target doxygen

avalenzu, 2018-04-03:
It'd be nice to not have to wait for `doxygen.py --quick` to run on the whole repo when I just changed comments in one file.

Assigning to @jwnimmer-tri for triage, because

[image]

SeanCurtis-TRI, 2018-04-03:
I'd accept some of those flags -- most of my recompiling is to make sure that things format the way I want them to. I can always do one last pass to make sure things are linking across drake correctly.

jwnimmer-tri, 2018-04-10:
@avalenzu or @SeanCurtis-TRI or anyone else -- any more thoughts on this?  Still seem like a good idea?

Assuming yes, what would be a convenient command-line interface.  Do you want to supply a list of filenames to process?  Or a cc_library label name or package name?  Or allow for globs like `solvers/*.h`?  Recursive globs `common/**/*.h`?

SeanCurtis-TRI, 2018-04-10:
I'll vote a big "yes".

 As for parameters, all of those options sound reasonable to varying degrees. I would want it tab-completable in the shell. But other than that, I can easily imagine circumstances in which any of those command-line parameters would be optimal (and, conversely, sub-optimal).

jwnimmer-tri, 2018-04-10:
Good point.  Since its a shell command line I should probably stick with shell semantics and filenames as the default.  If `doxgen.py` accepts a list of bare filenames (without `--some-switch=` ahead of them), then I think the shell's default glob expansion will do the right thing, and make the most sense to users.

jwnimmer-tri, 2018-05-24:
So it turns out that, yes, part of the slowness was generating documentation for lots of classes at once, and that narrowing the input helps a bit.

However, the bulk of the performance problem was that, in Doxygen if you set an `IMAGE_PATH`, on startup it immediately _recursively scans that path for all files, including following symlinks_, and because our image path has the `bazel-foo` symlinks at its root, it scans the entire Bazel cache, runfiles trees, workspaces, output tress, etc. to enumerate the possible image paths.  In my local build, that is about a million files.  This wasn't a problem in practice until #6996 moved the image path up one level.

I'll try to fix the image path scan as well as the input narrowing, in my next PR.

sherm1, 2018-05-24:
> scans the entire Bazel cache, runfiles trees, workspaces, output tress, etc. to enumerate the possible image paths

OMG no wonder it's been so slow!

PR #8867 (2018-05-24): Make Doxygen fast again
Change the way input files are handled.  Previously, we just pointed the tool at our workspace (git) root.  However, in this case Doxygen will recursively scan all of the bazel folders for IMAGE_PATH files, even though we have excluded the bazel folders as sources.  In local testing, this speeds up a --quick build from 93s to 23s.

Add an option to run over only certain input files or directories.  When specifying, e.g., systems/framework vs the whole tree, in local testing this speeds up a --quick build from 23s to 2s.

Fixes #8503.

---
This change is [[image]](https: [... cut]
closed 2018-05-24 by jwnimmer-tri