RobotLocomotion/drake #8504: Add is_any_of_nan / is_all_of_nan

soonho-tri, 2018-04-03:
feature request by @jwnimmer-tri (from Slack channel)

```c++
bool is_any_of_nan(const Eigen::Ref<MatrixX<Expression>>&);
bool is_all_of_nan(const Eigen::Ref<MatrixX<Expression>>&);
```

jwnimmer-tri, 2018-04-03:
I think we want them Formula-valued, and templated on T:
```c++
template <typename T>
drake::Bool<T> is_any_of_nan(const Eigen::Ref<MatrixX<T>>&);

template <typename T>
drake::Bool<T> is_all_of_nan(const Eigen::Ref<MatrixX<T>>&);
```

soonho-tri, 2018-04-03:
I'm wondering where we need to put those functions. Do you have a suggestion?

jwnimmer-tri, 2018-04-03:
I think `all_of` would likely only be used in unit tests -- to confirm that the "we guarantee to give you garbage" constructors actually do so.  Though in that case, `.array().isnan().all()` also works since we know there are no free variables.

soonho-tri, 2018-04-03:
FYI, I'm adding more generic `any_of` and `all_of` which takes a matrix *and* a predicate.

jwnimmer-tri, 2018-04-03:
New idea...

What if we mimic http://en.cppreference.com/w/cpp/algorithm/all_any_none_of with functions of that same name, but that take a `Matrix::Ref`, instead of iterators?  And place it in `drake_bool.h`?

soonho-tri, 2018-04-03:
@jwnimmer-tri OK. I'll add them in `drake_bool.h`.

soonho-tri, 2018-04-04:
Closed by #8508

closed 2018-04-04 by soonho-tri