RobotLocomotion/drake #8531: Unable to Generate Doxygen Website

liangfok, 2018-04-06:
I'm following the instructions [here](http://drake.mit.edu/documentation_instructions.html). Below is the console output:

```
$ ./doc/doxygen.py 
Building C++ Doxygen documentation... done
See file:///home/liang/dev/drake/build/drake/doc/doxygen_cxx/html/index.html

$ find . -name "*.html"
./build/drake/doc/doxygen_cxx/html/search/nomatches.html
./build/drake/doc/doxygen_cxx/html/index.html
./build/drake/doc/doxygen_cxx/html/graph_legend.html

$ doc/doxygen.py --quick
Building C++ Doxygen documentation... done
See file:///home/liang/dev/drake/build/drake/doc/doxygen_cxx/html/index.html

$ find . -name "*.html"
./build/drake/doc/doxygen_cxx/html/search/nomatches.html
./build/drake/doc/doxygen_cxx/html/index.html
./build/drake/doc/doxygen_cxx/html/graph_legend.html
```

Here's what I see when I open `index.html`:

[image]

I'm on Ubuntu 16.04.4 and ran `./setup/ubuntu/16.04/install_prereqs.sh`.

jwnimmer-tri, 2018-04-06:
What revision of Drake?  I assume something relatively recent?

It seems to work fine for other people.  Your output is consistent with no source code files being found.  If you look at `build/drake/doc/Doxyfile_CXX`, does the `DRAKE_WORKSPACE_ROOT` seem correct (`/home/liang/dev/drake`)?

liangfok, 2018-04-06:
I am using SHA 7f2a6bb, so it's relatively recent. When I open `build/drake/doc/Doxyfile_CXX`, I can't find a `DRAKE_WORKSPACE_ROOT`. I posted my Doxyfile_CXX here: https://gist.github.com/liangfok/e8be9a20f44e9b79e3d72285f2f5b124.

jwnimmer-tri, 2018-04-06:
Oops, I meant to say `INPUT` which is set to `DRAKE_WORKSPACE_ROOT`'s value.  Yours is `"/home/liang/dev/drake"` which seems correct.

What's the output of `which doxygen`?

liangfok, 2018-04-06:
Here's my output:
```
$ which doxygen
/usr/bin/doxygen
$ dpkg -l | grep doxygen
ii  doxygen                                                     1.8.11-1                                     amd64        Documentation system for C, C++, Java, Python and other languages
```

jwnimmer-tri, 2018-04-06:
Aha.  Look at `EXCLUDE_PATTERNS`.  A folder named `dev` anywhere on the path, even above the workspace root, will cull files.  (The intent is that _subdirectories_ within Drake named dev are ignored, but its too wide of a match.)

If you rename your checkout, it should start working.

jwnimmer-tri, 2018-04-06:
Drake needs to fix the pattern.

liangfok, 2018-04-06:
Oh OK. Thanks for opening it again.

PR #8532 (2018-04-08): Updates doxygen generation to allow */dev/* directories above DRAKE_WORKSPACE_ROOT.
Resolves #8531.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8532)
closed 2018-04-08 by jwnimmer-tri