RobotLocomotion/drake #8536: pydrake.symbolic.Formula and returns unexpected results

RussTedrake, 2018-04-08:
From underactuated piazza

``
mp = MathematicalProgram()
x = mp.NewContinuousVariables(1, 'x')
mp.AddLinearCost(x[0])
mp.AddLinearConstraint(-1 <= x[0] <= 1)
mp.Solve()
print mp.GetSolution(x)
```

This prints -6.17673396e+14 instead of printing -1 or throwing an exception. I understand that python doesn't allow you to override and, so AddLinearConstraint can only see one of the subexpressions (rhs if lhs is truthy, lhs if it isn't). But is there any reason why `pydrake._symbolic_py.Formula.__nonzero__` doesn't throw an exception?

soonho-tri, 2018-04-08:
I think #8500 resolves this issue as well. I'll check and add this as a test.

/cc @EricCousineau-TRI

EricCousineau-TRI, 2018-04-08:
Went ahead and added it in as a two-liner to #8500. Thanks!

commit a626f74f (2018-04-11): pydrake symbolic: Ensure compound formulas fail (Resolves #8536)
closed 2018-04-11 by soonho-tri