RobotLocomotion/drake #8544: Consider removing `@numpy` repository

EricCousineau-TRI, 2018-04-08:
From what I understand, `pybind11` does not need `numpy` headers as it a separate ABI (not through `ld.so`), which effectively "links" the functions via dictionaries in Python that store the relevant pointers in capsules, and locks constants by an API guarantee in the documentation.

Because of this, we don't really need `@numpy` as a C++ dependency; it's existence is effectively guaranteed by `install_prereqs.sh`.

@rdeits Can I ask if there was any reason in particular for keeping these headers around? Since @m-chaturvedi submitted an updated patch which uses the above ABI style, it shouldn't be needed for `dtype=object` stuff.

rdeits, 2018-04-09:
Nope, no reason I'm aware of.

jamiesnape, 2018-04-09:
> Because of this, we don't really need @numpy as a C++ dependency; it's existence is effectively guaranteed by install_prereqs.sh.

We should not be using `install_prereqs.sh` to guarantee anything. Presumably we will have failing tests in Drake, which is the guarantee.

PR #14116 (2020-09-24): Deprecate the numpy repository
Closes #8544.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14116)
commit 1fc03da8 (2020-09-24): Deprecate the numpy repository (#14116)

Closes #8544.
closed 2020-09-24 by sammy-tri