RobotLocomotion/drake #8556: == vs <=, >= in nonlinear constraints

dangthatsright, 2018-04-09:
Hi, I am in Russ' underactuated robotics class and I was told I should post this here. 

In our problem set, we are simulating a system with two worlds and a rocket. I specify an initial condition and impose an end condition. This code uses direct transcription to find a valid path. 
```
from numpy import sin, cos
import numpy as np

from pydrake.all import MathematicalProgram

class OrbitalTransferRocket():

    def __init__(self):
        self.G  = 9.8  # gravitational constant
        self.M1 = 0.4  # mass of first planet
        self.M2 = 0.1  # mass of second lanet
        self.world_1_position = np.asarray([-2.5,-0.1])
        self.world_1_color    = "black"   # change to 'darkgreen' for photorealistic Earth sim
        self.world_2_position = np.asarray([2.5,0.1])
        self.world_2_color    = "black"   # change to 'orangered' for photorealistic Mars sim

    def rocket_dynamics(self, state, u):
        '''
        Calculates the dynamics, i.e.:
           \dot{state} = f(state,u)

        for the rocket + two planets system.

        :param state: numpy array, length 4, comprising state of system:
            [x, y, \dot{x}, \dot{y}]
        :param u: numpy array, length 2, comprising control input for system:
            [\ddot{x}_u, \ddot{y}_u]   
            Note that this is only the added acceleration, note the total acceleration.

        :return: numpy array, length 4, comprising the time derivative of the system state:
            [\dot{x}, \dot{y}, \ddot{x}, \ddot{y}]
        '''

        rocket_position = state[0:2]
        derivs = np.zeros_like(state)
        derivs[0:2] = state[2:4]

        # these local copies of variables make the equations below
        # easier to look at
        G = self.G; M1 = self.M1; M2 = self.M2;
        world_1_position = self.world_1_position; world_2_position = self.world_2_position;

        derivs[2]  = G * M1 * (world_1_position[0] - rocket_position[0]) / self.two_norm(world_1_position - rocket_position)**3
        derivs[2] += G * M2 * (world_2_position[0] - rocket_position[0]) / self.two_norm(world_2_position - rocket_position)**3
        derivs[2] += u[0]
        derivs[3]  = G * M1 * (world_1_position[1] - rocket_position[1]) / self.two_norm(world_1_position - rocket_position)**3
        derivs[3] += G * M2 * (world_2_position[1] - rocket_position[1]) / self.two_norm(world_2_position - rocket_position)**3
        derivs[3] += u[1]
        
        return derivs

    def two_norm(self, x):
        '''
        Euclidean norm but with a small slack variable to make it nonzero.
        This helps the nonlinear solver not end up in a position where
        in the dynamics it is dividing by zero.

        :param x: numpy array of any length (we only need it for length 2)
        :return: numpy.float64
        '''
        slack = .001
        return np.sqrt(((x)**2).sum() + slack)

    def compute_trajectory_to_other_world(self, state_initial, minimum_time, maximum_time):
        '''
        :param: state_initial: :param state_initial: numpy array of length 4, see rocket_dynamics for documentation
        :param: minimum_time: float, minimum time allowed for trajectory
        :param: maximum_time: float, maximum time allowed for trajectory

        :return: three return args separated by commas:

            trajectory, input_trajectory, time_array

            trajectory: a 2d array with N rows, and 4 columns. See simulate_states_over_time for more documentation.
            input_trajectory: a 2d array with N-1 row, and 2 columns. See simulate_states_over_time for more documentation.
            time_array: an array with N rows. 

        '''
        # initialize math program
        mp = MathematicalProgram()
        num_time_steps = 50
        total_time = 5.0
        dt = total_time/num_time_steps

        time_array = np.arange(0.0, total_time, dt)

        booster = mp.NewContinuousVariables(2, "booster_0")
        booster_over_time = booster

        state = mp.NewContinuousVariables(4, "state_0")
        state_over_time = state

        for k in range(1,num_time_steps):
            booster = mp.NewContinuousVariables(2, "booster_%d" % k)
            booster_over_time = np.vstack((booster_over_time, booster))
            state = mp.NewContinuousVariables(4, "state_%d" % k)
            state_over_time = np.vstack((state_over_time, state))

        # calculate states over time
        for i in range(1,len(time_array)):
            time_step = time_array[i] - time_array[i-1]
            sim_next_state = state_over_time[i-1,:] + time_step*self.rocket_dynamics(state_over_time[i-1,:], booster_over_time[i-1,:])
            
            # add constraints to restrict the next state to the decision variables
            mp.AddConstraint(sim_next_state[0] >= state_over_time[i,0])
            mp.AddConstraint(sim_next_state[0] <= state_over_time[i,0])
            mp.AddConstraint(sim_next_state[1] >= state_over_time[i,1])
            mp.AddConstraint(sim_next_state[1] <= state_over_time[i,1])
            mp.AddConstraint(sim_next_state[2] >= state_over_time[i,2])
            mp.AddConstraint(sim_next_state[2] <= state_over_time[i,2])
            # REPLACE ABOVE TWO CONSTRAINTS WITH: mp.AddConstraint(sim_next_state[2] == state_over_time[i,2])
            mp.AddConstraint(sim_next_state[3] == state_over_time[i,3])

        # add constraints on mp_state_initial so they are fixed to state_initial
        mp.AddLinearConstraint(state_over_time[0,0] == state_initial[0])
        mp.AddLinearConstraint(state_over_time[0,1] == state_initial[1])
        mp.AddLinearConstraint(state_over_time[0,2] == state_initial[2])
        mp.AddLinearConstraint(state_over_time[0,3] == state_initial[3])

        # add constraints on end state
        position_goal = np.asarray([2.5, 0.4, 1.0, 0])

        mp.AddConstraint(state_over_time[-1,0] >= position_goal[0])

        # add the cost function
        mp.AddQuadraticCost(1 * booster_over_time[:,0].dot(booster_over_time[:,0]))
        mp.AddQuadraticCost(1 * booster_over_time[:,1].dot(booster_over_time[:,1]))

        mp.Solve()

        booster_over_time = mp.GetSolution(booster_over_time)
        output_states = mp.GetSolution(state_over_time)

        return output_states, booster_over_time, time_array
```

which can be run with the following initial conditions
```
rocket = OrbitalTransferRocket()

x0 = np.array([-2., 0.1, -0.1, 3])
minimum_time = 5.
maximum_time = 15.
traj, u_traj, time_array = rocket.compute_trajectory_to_other_world(x0, minimum_time, maximum_time)
print u_traj
```
If you look above and replace 
```
mp.AddConstraint(sim_next_state[2] >= state_over_time[i,2])
mp.AddConstraint(sim_next_state[2] <= state_over_time[i,2])
```
with 
```
mp.AddConstraint(sim_next_state[2] == state_over_time[i,2])
```
the former gives us a pretty reasonable result, but the latter gives us an all 0 control. 

I know there are discrepancies between == and >=, <= for nonlinear constraints but the differences between these two results seem unreasonable.

hongkai-dai, 2018-04-09:
There is difference between an equality constraint `f(x) == a` from inequality constraint `a <= f(x) <= a`. In the former case, It is treated by the solver as one equality constraint, the latter case is treated by the solver as two inequality constraint `a <= f(x) <= infinity` and `-infinity <= f(x) <= a`. The solver tries to solve the nonlinear nonconvex problem iteratively. In each iteration, it needs to determine which constraint shall be active (by active I mean to take equality). For an equality constraint, it is always active; for inequality constraint, there is some additional numerical procedure to determine whether to activate the inequality constraint. Due to the difference in treating equality and inequality constraint, in each iteration the result can be very different, and thus leading to the different result in the end of the optimization. 

The gist is to always use equality constraint `f(x) == a` instead of two inequality constraint `a <= f(x) <= a`.

peteflorence, 2018-04-09:
Hey Hongkai, agreed, but the issue is that two inequality constraints works
for this problem, while an equality constraint gives a non-sensible
answer... that's the confusing part.

hongkai-dai, 2018-04-09:
Which solver is used? SNOPT or IPOPT?

peteflorence, 2018-04-09:
SNOPT

peteflorence, 2018-04-10:
I edited this comment so that it didn't contain any answers to this week's problem set.

jwnimmer-tri, 2018-05-01:
@hongkai-dai At some point could you investigate and see if this is a defect in Drake?