RobotLocomotion/drake #8557: Labels (names) for python tests are confusing

jwnimmer-tri, 2018-04-09:
Bazel's test output looks like this (taken from CI, for example):
```
//bindings/pydrake/examples:_isolated/acrobot_test                       PASSED in 6.1s
//bindings/pydrake/examples:acrobot_test                                 PASSED in 6.1s
```
However, this is really just one test (its `alias`ed).

The problem is that only one of the labels is actually usable for developers:
```console
jwnimmer@call-cps:~/jwnimmer-tri/drake$ bazel test //bindings/pydrake/examples:acrobot_test
...
ERROR: No test targets were found, yet testing was requested

jwnimmer@call-cps:~/jwnimmer-tri/drake$ bazel test //bindings/pydrake/examples:_isolated/acrobot_test
...
INFO: Found 1 test target...
...
Executed 1 out of 1 test: 1 test passes.
```

Having labels that don't work seems unhelpful.  Can we remove the `alias()` calls?

EricCousineau-TRI, 2018-04-11:
At a first glance, the underlying issues are that (a) Bazel doesn't recognize aliases via `bazel test` and (b) CI does not group `alias`ed tags (granted, it stinks that the Bazel does not consolidate these before the fact).

One workaround for running the test directly is to use `bazel run` in lieu of `bazel test`, as it is able to recognize that the target is an `alias`:
```
$ bazel run //bindings/pydrake/examples:acrobot_test
...
Ran 4 tests in 0.004s

OK
```
Unfortunately, this still means that you cannot add this to a list of explicit targets to be tested, and you do not get cached results / condensed output.

My main concern with removing the `alias` is that any downstream components that refers to the test component must now refer to what was intended to be an implementation detail.

Brainstorming some potential solutions:
* Keep things as-is (not ideal when wanting to use `bazel test`, and reduce clutter on CI output)
* Remove `alias`ed portions from `drake_py_*test` - consider changing the prefix from `_isolated/` to something like `_py/` or just `py/` - still, not ideal.
* Address the original issue for `_isolated/`: for any package that may introduce shadowed names, ensure that any explicitly defined tests (e.g. not linting), ensure that the tests are defined in a separate folder.
* Bake any shadowy names into a top-level module, or do some shifty magic to try and separate the module definition from the folder layout.

jwnimmer-tri, 2018-04-11:
> Unfortunately, this still means that you cannot add this to a list of explicit targets to be tested, and you do not get cached results /  condensed output.

Plus the test environment is totally different (https://docs.bazel.build/versions/master/test-encyclopedia.html#test-interaction-with-the-filesystem and such).

> My main concern with removing the alias is that any downstream components that refers to the test component must now refer to what was intended to be an implementation detail.

Can you give an example of this?  What depends on test rules?

My first impression of solutions are:
- Best to just kill the alias, and live with the gross names.  This is correct but unaesthetic.  Oh well.
- Stop using "_isolated" idiom entirely somehow, likely by no longer using shadowy names.  It seems like an uphill battle.

jwnimmer-tri, 2018-04-30:
Even week or two, somebody stumbles on this.  Bumping to "priotity: high".  What's stopping us from just nuking the lying aliases for now?

EricCousineau-TRI, 2018-04-30:
Making the PR now; however, autocomplete will still report the incorrect alias. If anything, I believe that we'll still get questions about why both `bazel test` and `bazel run` don't work.

I don't like it, but if you think it'll improve the situation and since I don't have a better suggestion, I'm up for trying it out.

PR #8724 (2018-05-01): drake_py: Remove alias from isolated targets; rename prefix from `_isolated/` to `py/`
Resolves #8557

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8724)
closed 2018-05-01 by sammy-tri