RobotLocomotion/drake #8559: Get Drake to output trajectory/ state evolution

AlameluThinnappan, 2018-04-10:
For my research, I would like to import URDF files of robots, and get Drake to output controllers and state evolutions for planned motions. The existing examples publish states to the simulator. It would be great to have some direction here on what parts of the code have to be modified to print out state evolutions. 
Thank you!

RussTedrake, 2018-05-02:
I'm not very clear on what you need, but perhaps you want to look at our uses of [PlanEval](http://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1controllers_1_1plan__eval_1_1_plan_eval_base_system.html#details)?

RussTedrake, 2018-05-02:
Will close this for now.  For general questions, please consider our newly advertised tag on StackOverFlow (http://drake.mit.edu/getting_help.html)

closed 2018-05-02 by RussTedrake