RobotLocomotion/drake #8563: Add retry logic to license acquisition in GurobiSolver

jamiesnape, 2018-04-10:
Since it is common for the license to be obtained from a license server over a network, it is perfectly possible that license acquisition will fail on occasion.

jamiesnape, 2018-04-11:
Interim solution to license server flakiness is to add the flag `--flaky_test_attempts=3` build option on CI for "everything" builds.

mpetersen94, 2018-07-11:
Additionally, every time the GurobiSolver is used to solve a mathematical program, a message is printed if using a Gurobi academic license.  If the GRBenv (or whatever is causing the re-printing) could be saved to limit the number of messages to 1 that would be very helpful (something similar to `MosekSolver::AcquireLicense`).

PR #9126 (2018-07-19): Make Gurobi retry downloading license
Closes #8563

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9126)
closed 2018-07-19 by jwnimmer-tri