RobotLocomotion/drake #8576: Usage of python terminal inside drake visualizer?

CatchMeFastFat, 2018-04-11:
What is the usage of python terminal inside drake visualizer?

I thought that it would be  https://cdn.rawgit.com/root-project/cling/master/www/index.html cling"c++ interpreter"

I know it would help in director, but what is the use case for drake alone?

EricCousineau-TRI, 2018-04-11:
May I ask what sort of workflows you wish to accomplish with `drake-visualizer`, and with `drake` in general?

`pydrake` / `drake` alone is meant for doing the core numeric portions, and then communicating with other components (`matplotlib` or `drake-visualizer`) to do the visualization.

`drake-visualizer` is meant for visualization of simulation scenes (robots, point clouds, etc.), but it's not necessarily meant for running simulations using `pydrake` - you generally launch `drake-visualizer` separately, and then use IPC from `pydrake` to draw.

You may want to consider using something like Jupyter / IPython notebooks, so that you can more easily save your work (and visualize plots, etc).
If you're gearing towards interactive plots, I believe @avalenzu may have messed around with interactive plots in IPython, specifically for manipulating a robot being visualized in `drake-visualizer`.

W.r.t. `cling`, I had briefly tested out `cling` in from a Jupyter kernel using [this article](https://blog.jupyter.org/interactive-workflows-for-c-with-jupyter-fe9b54227d92?gi=e0f762988409), but I had only tinkered with Eigen, and did not try linking to Drake.

My suggestion would be to use `pydrake` from IPython / Jupyter (and certainly submit feature requests / patches if there's gaps in what you need); if you need something closer to bare metal, then programming in C++ would be best, as `cling` itself may be a tad slow, and may have some caveats on what it can parse, even using an LLVM backend, as we use some of the fancier C++1y features. However, if you do get it working well, please do let us know!

CatchMeFastFat, 2018-04-11:
OK. I will try cling. 

Do you have any plans to update http://drake.mit.edu/doxygen_cxx/index.html? if not I think it would help to add http://drake.mit.edu/documentation_instructions.html#documentation-generation-instructions to install instructions.

CatchMeFastFat, 2018-04-15:
@EricCousineau-TRI 

Cling seems to work.

here is what I have tried:

got cling-ubuntu16 https://root.cern.ch/download/cling/ 
`export PATH=$PATH:/home/drake/cling_2018-04-11_ubuntu16/bin/:/home/drake/cling_2018-04-11_ubuntu16/include:/home/drake/cling_2018-04-11_ubuntu16/lib`

started drake-visualizer
then cling  with the following flags:

```
$ cling -std=c++14 -L../../build/install/drake/lib \
-I../../build/install/drake/include -I../../build/install/drake/include/stx \
-I../../build/install/drake/include/eigen3 \
-I../../build/install/drake/include/lcm \
-I../../build/install/drake/include/drake_lcmtypes \
-llibdrake -llibgflags -llibdrake_lcm
```
`__cplusplus` inside cling gives c++ standard used.
I used double_pendulum example only"others have multiple sources".

```
#include <memory>
#include <gflags/gflags.h>
#include <sys/stat.h>
#include "drake/common/find_resource.h"
#include "drake/common/text_logging_gflags.h"
#include "drake/lcm/drake_lcm.h"
#include "drake/multibody/joints/floating_base_types.h"
#include <drake/multibody/parsers/sdf_parser.h>
#include "drake/multibody/rigid_body_plant/drake_visualizer.h"
#include "drake/multibody/rigid_body_plant/rigid_body_plant.h"
#include "drake/systems/analysis/simulator.h"
#include "drake/systems/framework/diagram.h"
#include "drake/systems/framework/diagram_builder.h"
#include "drake/systems/primitives/constant_vector_source.h"

#include<iostream>

/// Fixed path to particle SDF model (for visualization purposes only).
static const char* const kPendulumSdfPath = "double_pendulum.sdf";

using namespace drake;
// lcm without _ gives error: redefinition of 'lcm' as different kind of symbol drake::lcm::DrakeLcm lcm;
drake::lcm::DrakeLcm lcm_;

auto tree = std::make_unique<RigidBodyTree<double>>();

parsers::sdf::AddModelInstancesFromSdfFileToWorld(kPendulumSdfPath, multibody::joints::kFixed, tree.get());

tree->compile();

// Instantiate builder.
systems::DiagramBuilder<double> builder;

// Move double pendulum tree to plant.
auto plant = builder.template AddSystem<systems::RigidBodyPlant<double>>(
    std::move(tree));

// Add visualizer client.
auto visualizer = builder.AddSystem<systems::DrakeVisualizer>(plant->get_rigid_body_tree() , &lcm_);

// Wire all blo`cks together.
builder.Connect(plant->get_output_port(0), visualizer->get_input_port(0));
auto diagram = builder.Build();
// Instantiate and configure simulator.

systems::Simulator<double> simulator(*diagram);
// auto simulator = std::make_unique<systems::Simulator<double>>(*diagram);
simulator.set_target_realtime_rate(1.0);
simulator.Initialize();

// Run simulation.
simulator.StepTo(10);
```
Speed wasn't bad only `auto diagram = builder.Build();` took noticeable time around 2-3 mins.

EricCousineau-TRI, 2018-05-01:
Sorry for the delayed response! That's awesome that you got it working!

I'm not sure we're to the point where we want to support this ourselves, but will consider mentioning it on the documentation if we have more requests for it.

Given that the current issue's title is geared towards `drake_visualizer`, would you mind if I closed the current issue?
Or would you like to rename the title to something like `Add example Jupyter notebook using cling C++ interpreter`? (may still knock it down on priority, though)

EricCousineau-TRI, 2018-05-15:
Thanks!

> Could that be faster?

Perhaps, but unfortunately I'm not aware of anything to improve the speed. There may be things you can tweak in `cling`, but that'd be something that you'd need to ask the authors of that tool.

> should I use another lighterweight visualizer?

I'd suggest using `pydrake`, as `DrakeVisualizer` is already available. An example (via a unittest): (sorry that I hadn't pointed this example out earlier!)
https://github.com/RobotLocomotion/drake/blob/918d7017e36c689bef716d1a89d3a076052bf60e/bindings/pydrake/multibody/test/rigid_body_plant_test.py#L214-L231

If there is anything missing off of the current `master` that you need for `pydrake`, please do consider opening another issue as a feature request.

jwnimmer-tri, 2018-09-27:
This seems resolved.  Please re-open with more info, in case not.

closed 2018-09-27 by jwnimmer-tri