RobotLocomotion/drake #8589: Relatively tight thetadot tolerance causes rimless wheel test to break

edrumwri, 2018-04-12:
Pinging @RussTedrake:

If we change the tolerance for next_state.thetadot() from 0.01*x, where x = sqrt(params.gravity()/params.length()) to tol*x [Line 108 of rimless_wheel.cc] and -.01*x to -tol*x [Line 161 of rimless_wheel.cc], the test returns the following output:

==================== Test output for //examples/rimless_wheel:rimless_wheel_test:
Using drake_cc_googletest_main.cc

[==========] Running 3 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 3 tests from RimlessWheelTest
[ RUN      ] RimlessWheelTest.ToAutoDiff
[       OK ] RimlessWheelTest.ToAutoDiff (4 ms)
[ RUN      ] RimlessWheelTest.StepTest
[       OK ] RimlessWheelTest.StepTest (20 ms)
[ RUN      ] RimlessWheelTest.FixedPointTest
abort: Failure at examples/rimless_wheel/rimless_wheel.cc:92 in StepForwardReset(): condition 'next_state.theta() > params.slope() - alpha' failed.

Debugging indicates that next_state.theta() = params.slope() - alpha. I have not identified an elegant fix to this problem.

Note: I came across this while trying to replace the constant 0.01 with the accuracy value set in the context.

sherm1, 2018-04-12:
I volunteer to take a crack at sorting this out, with @edrumwri as my consultant. I feel that it should be relatively easy to make this system completely bulletproof in a straightforward way and I'm interested to see where we're falling short. Assigned to me.

RussTedrake, 2018-04-12:
i know what it is.  you've changed the number of steps the wheel takes before it enters double support.  so theta = slope+alpha now instead of slope-alpha.  

the reason I didn't change the test to allowing either slope +/- alpha before was because evan pointed out that I was missing code coverage on the transition to double support when I didn't have one test case hitting each of the triggers.

jwnimmer-tri, 2022-05-01:
@sherm1 or @RussTedrake is this issue still important?  (It being dormant for 4 years suggests maybe not.)  I wonder if we can add a TODO comment (or similar) to the example code to capture the thoughts here, and then close this out without any implementation changes?

PR #17078 (2022-05-03): [examples/rimless_wheel] Resolves subtle numerics in witness function logic
Resolves #8589. If the guard was triggered with exactly zero
penetration, then we could fail on the following step because the
wheel was positioned exactly in contact with the ground (and we are
using WitnessFunctionDirection::kPositiveThenNonPositive). The
unfortunate solution is to add machine epsilon to make the next
witness value strictly positive.

This does *not* implement the change to the double support threshold
suggested in #8589, where we use the accuracy from the context as the
threshold for detecting doubl [... cut]
commit 66727da1 (2022-05-03): [examples/rimless_wheel] Resolves subtle numerics in witness function logic (#17078)

* [examples/rimless_wheel] Resolves subtle numerics in witness function logic.

Resolves #8589. If the guard was triggered with exactly zero
penetration, then we could fail on the following step because the
wheel was positioned exactly in contact with the ground (and we are
using WitnessFunctionDirection::kPositiveThenNonPositive). The
unfortunate solution is to add machine epsilon to make the next
witness value strictly positive.

This does *not* implement the change to the double support threshold
suggested [... cut]
closed 2022-05-03 by RussTedrake