RobotLocomotion/drake #8604: Add DRAKE_DEMAND to `to_variable` and its variants in symbolic

soonho-tri, 2018-04-16:
From https://github.com/RobotLocomotion/drake/pull/8600#issuecomment-381726312

jwnimmer-tri, 2018-04-16:
We already have `DRAKE_ASSERT(is_variable(...))` etc in all of those.  Is the proposal to promote to `DRAKE_DEMAND` because developers will make too many hard-to-find errors otherwise?

jwnimmer-tri, 2018-04-16:
To my eye, the problem with #8600 is that probably that new routine should have a total contract, not a precondition -- that it should check `is_variable` in the loop and throw a useful error message if not, instead of relying on the bare-metal conversion functions to guard and report useful errors.

soonho-tri, 2018-04-17:
> that new routine should have a total contract, not a precondition -- that it should check is_variable in the loop and throw a useful error message if not, instead of relying on the bare-metal conversion functions to guard and report useful errors.

It makes sense to make `get_variable_vector` a total function. Then, to be consistent, we need to make the other existing symbolic getters such as `get_variable` and `get_constant_value` total functions as well (for now, they are partial functions with pre-conditions). @jwnimmer-tri , do you agree?

Originally, I wanted to avoid making them total function because of the checking overheads. N.B. they're used frequently in symbolic manipulation.

/cc @RussTedrake and @hongkai-dai (authored and reviewed #8600)

jwnimmer-tri, 2018-04-17:
> It makes sense to make get_variable_vector a total function.

I agree.  Possibly this should not be a lowercase function either, since _Such methods should have a time complexity of O(1) and be less than 5 lines long._ does not apply here (linear complexity).

> Then, to be consistent, we need to make the other existing symbolic getters such as get_variable and get_constant_value total functions as well (for now, they are partial functions with pre-conditions). @jwnimmer-tri , do you agree?

I do not agree.  I perceive those as high-performance, low-level casting functions for use by developers would build sugar such as `get_variable_vector`, or implement symbolic primitives, or etc.  I do not see the need for consistency.

> Originally, I wanted to avoid making them total function because of the checking overheads. N.B. they're used frequently in symbolic manipulation.

I agree.

soonho-tri, 2018-04-17:
Thanks for all the comments. I'll rename `get_variable_vector` to `GetVariableVector`, make it a total function, and update the documentation.

jwnimmer-tri, 2018-04-17:
(In my research for #5268, I have a few whitepapers about this that I will summarize for the group soon, including a probably for a consistent Drake-wide way to think about these questions, and phrase them into runtime code.  There is a John Lakos talk / whitepaper on defensive programming that I am quite enjoying.)

soonho-tri, 2018-04-17:
Interesting. Could you post them in the DrakeDeveloper Slack team (such as #cpp or #general)?

PR #8614 (2018-04-17): Rename get_variable_vector to GetVariableVector and make it a total function
Also make it a total function.

Close #8604

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8614)
closed 2018-04-17 by soonho-tri