RobotLocomotion/drake #8610: Stop using `exclusive` tag on Gurobi tests

jamiesnape, 2018-04-17:
At least on CI, since it wastes resources and impacts remote caching as it is a very blunt instrument that causes test to run sequentially regardless of whether parallel tests would actually be using Gurobi. Appears non-trivial at first glance, however.

jamiesnape, 2018-07-16:
One possibility would be to slightly abuse the `large` test size. The number of `large` tests that may run in parallel is limited by Bazel.

jwnimmer-tri, 2020-05-14:
In CI, is it the case that we have unlimited license concurrency?  So only developer's own builds might have a concurrency limit given that local setup?

We could add a CI opt-out on the limit, by changing the `gurobi_test_tags()` macro.

Another way to limit concurrency is `tags = ["cpu:999"]` which will run at most two gurobi tests concurrently.  That would at least step away from "exclusive".

jamiesnape, 2020-05-14:
We have 4096 licenses, which would allow for ~256 concurrent builds, which is more than our AWS instance quota, so essentially unlimited. We can also enforce the maximum number of builds that use Gurobi on Jenkins, if necessary.

So to answer the question, it is probably only commercial developer builds that run into issues.

jwnimmer-tri, 2021-11-11:
The solution here will be a PR to alter the `gurobi_test_tags()` macro so that a bazel command line option (e.g., `--define=GUROBI_EXCLUSIVE_TESTS=OFF`) will opt-out of the `"exclusive"` tag, but leaves the current defaults intact.  Then in CI (only), we will turn on that definition.  (CI is the only place that has unlimited licenses, most all users will always need a throttle.)

jwnimmer-tri, 2023-01-18:
This would be worth doing.  As it stands now, Gurobi tests in CI (especially the `mixed_integer_rotation_constraint_test`) are not being cached ("exclusive" turns off caching), and are frequently the only thing holding up PR merging for like 10 minutes of extra waiting.

svenevs, 2023-01-23:
We are not clear on how to test any of this, other than to just remove `exclusive` and let it run and hope it works?  How else are we going to try to max out the license concurrency?

jwnimmer-tri, 2023-01-23:
I'm not exactly sure what you're asking

In terms of overloading CI, per the above comments there's not any real risk of that.  (If we see it happening, we'll just revert the PRs.)  No need to test that.

Is the question how to tell whether the `bzl` logic is correctly adding the flag to local builds, but not adding it to CI builds?

For proving that local builds are unaffected, I think that running `bazel test //some_gurobi_test1 //some_gurobi_test2` should be enough see if they are running serially (desired) or parallel (undesired for local builds).

For proving that CI builds now run in parallel, it might be more difficult to test but anyway we can experiment locally adding the same flags a CI, and watch it run in parallel.

jwnimmer-tri, 2023-01-24:
Hmm, I'd forgotten that `tags = ...` is not configurable, so this will require slightly more work than just a `--define=GUROBI_EXCLUSIVE_TESTS=OFF`.  Let me prototype something and get back to you.

jwnimmer-tri, 2023-01-24:
Here's a rough outline of the build system patch.  I haven't tested it at all, but it should be enough of an outline to hopefully get you going.

For CI, we'd do `export GUROBI_EXCLUSIVE=0` in order to run tests concurrently.  Users would not need to change anything, since we default the environment variable to "1" in this patch.

```patch
--- a/tools/skylark/test_tags.bzl
+++ b/tools/skylark/test_tags.bzl
@@ -6,6 +6,8 @@
 # for any license-related needs and provide a marker so that //tools/bazel.rc
 # can selectively enable tests based on the developer's chosen configuration.
 
+load("@gurobi//:defs.bzl", "GUROBI_EXCLUSIVE")
+
 def gurobi_test_tags(gurobi_required = True):
     """Returns the test tags necessary for properly running Gurobi tests.
 
@@ -13,19 +15,24 @@ def gurobi_test_tags(gurobi_required = True):
     tag filters include "gurobi".
 
     Gurobi checks a license file outside the workspace so tests that use Gurobi
-    must have the tag "no-sandbox". For the moment, we also require the tag
-    "exclusive" to rate-limit license servers with a small number of licenses.
-    """
+    must have the tag "no-sandbox".
 
-    # TODO(david-german-tri): Find a better fix for the license server problem.
-    nominal_tags = [
-        "exclusive",  # implies "local"
+    Unless GUROBI_EXCLUSIVE=0 is set in the shell environment (e.g., in CI),
+    we also require the tag "exclusive" to rate-limit license servers with a
+    small number of licenses.
+    """
+    result = [
+        # TODO(david-german-tri): Find a better fix for the license file.
         "no-sandbox",
     ]
+
+    if GUROBI_EXCLUSIVE:
+        result.append("exclusive")
+
     if gurobi_required:
-        return nominal_tags + ["gurobi"]
-    else:
-        return nominal_tags
+        result.append("gurobi")
+
+    return result
 
 def mosek_test_tags(mosek_required = True):
     """Returns the test tags necessary for properly running MOSEK™ tests.
diff --git a/tools/workspace/gurobi/repository.bzl b/tools/workspace/gurobi/repository.bzl
index 4e94717761..9cf6fcd6e8 100644
--- a/tools/workspace/gurobi/repository.bzl
+++ b/tools/workspace/gurobi/repository.bzl
@@ -67,8 +67,17 @@ def _gurobi_impl(repo_ctx):
         executable = False,
     )
 
+    # Capture whether or not Gurobi tests can run in parallel.
+    exclusive_int = repo_ctx.os.environ.get("GUROBI_EXCLUSIVE", "1")
+    is_exclusive = bool(int(exclusive_int) != 0)
+    repo_ctx.file(
+        "defs.bzl",
+        content = "GUROBI_EXCLUSIVE = {}".format(is_exclusive),
+        executable = False,
+    )
+
 gurobi_repository = repository_rule(
-    environ = ["GUROBI_HOME"],
+    environ = ["GUROBI_HOME", "GUROBI_EXCLUSIVE"],
     local = True,
     implementation = _gurobi_impl,
 )
```

svenevs, 2023-01-24:
Thank you for the clarifications / starter code!  Indeed I was a bit confused about what we were testing here, but we understand what the end game scenario looks like with local testing and expected concurrency changes depending on the value of `GUROBI_EXCLUSIVE`.  We will be using this as an excuse to get @williamjallen oriented on how to deploy / use a CI machine to test this (only the CI machines can use the gurobi license server).

closed 2023-02-08 by rpoyner-tri