RobotLocomotion/drake #8616: Documentation for how to write a leaf system

jwnimmer-tri, 2018-04-17:
We should add documentation for how to write a leaf system, for example including a decision tree about the system model you are choosing, whether / how to do symbolic and autodiff, abstract vs numeric state, etc.  We have various examples of varying quality, but no tutorial that stitches them together.

sherm1, 2018-04-17:
/cc @ManuelAhumada-TRI as a friendly guinea pig for this documentation.

ManuelAhumada-TRI, 2018-04-17:
Sure thing! I'd be in to test it (i.e. guinea pig).

edrumwri, 2019-12-19:
One piece of information to be attached to this issue: we need documentation for declaring events and writing event handlers for LeafSystem authors.

jwnimmer-tri, 2024-11-01:
Fixed as of the `authoring_leaf_systems.ipynb` tutorial.

closed 2024-11-01 by jwnimmer-tri