RobotLocomotion/drake #8637: Need Diagram-level event handlers for discrete MBPlant/GeometrySystem

sherm1, 2018-04-19:
@amcastro-tri ran into this problem trying to make a discrete version of MultibodyPlant: in order to perform the discrete update, which includes an update to q, he wanted to evaluate the end-of-step penetration distance. But now the geometric information is in another subsystem. The MBPlant handler he is in only gets Context for MBPlant; it doesn't include the Context for GeoSystem.

We don't have this problem with RBPlant because it includes the geometry, and we don't have a problem when the Simulator advances the state (using an integrator) because it has access to the full Diagram context.

IMO this is another example like #4114 where it is clear that a Diagram should have the same superpowers we grant to a LeafSystem -- we should be able to define Diagram level event handlers at least (use cases for Publish and DiscreteUpdate have come up), and reconsider why we don't give Diagrams full resource allocation capability (ports, state variables, etc.).

SeanCurtis-TRI, 2018-04-19:
As I understand it, this is insufficient.

1. `GeometryState` *has* an output port that is connected to `MBP`.
2. That output port, in turn, depends on an `MBP` output port (and therefore the state in its context).
3. In MBP's discrete event handler, it wants to get information from `GeoemtryState` (through its port) for *various* MBP state configurations.
4. However, it has no write access to the continuous state, so `GeometryState` will *always* give the same answer based on the continuous state in the context on which the discrete event handler was called. MBP can't use it to perform queries on perturbations of its state. In this case, the "perturbation" was a speculative future state based on current configuration and velocity.

sherm1, 2018-04-19:
This wouldn't be giving MBPlant access to GeoSys's context. Instead there would be a Diagram (a kind of System), and it would have access only to its own context. That Diagram is a composition of MBPlant and GeoSys so can make speculative changes that affect both without breaking down any walls.

We have exactly this situation in the Simulator already. It works entirely with a Diagram and its Context. It performs speculative execution by modifying the Diagram's context, calling EvalDerivatives or whatever, and responding to the result. No dependency-tracking is harmed -- it's just that the work is getting done at the level where the appropriate information is available.

jwnimmer-tri, 2018-04-19:
@SeanCurtis-TRI In that case MultibodyPlant needs some API by which it does the relevant hypothetical computations (assuming it can't just do them internally).  It already has https://github.com/RobotLocomotion/drake/blob/7cc7d43/multibody/multibody_tree/multibody_plant/multibody_plant.cc#L80 a pointer to the geometry_system_.  What else does it need?  If it needs more information from GeometrySystem than is currently passed through the output port, then add it to that output (or a new output).

SeanCurtis-TRI, 2018-04-19:
As @sherm1 points out, it's all about access to context. GeometrySystem will allow every possible query. But every query it answers will be a function of its context. If the context doesn't change, the same query will produce the same answer -- based on whatever state is included in the GeoemetrySystem's context (and its parent Diagram's context). 

The implication of what you're suggesting is to provide an API to GeometrySystem that subverts the context. All of the inputs that would ordinarily come in via a port can be fed directly and a result would come out. However, GeometrySystem stores most of its important structures in its *own* context. So, it would end up thrashing its own context.

We can solve this, but I want to emphasize that I don't see any solution is a "just do A" kind of solution.

Perhaps @edrumwri can weigh in, but I assume that the only way he can do RBP time stepping is because the RBT *owns* the collision model. N'cest pas?

jwnimmer-tri, 2018-04-19:
@SeanCurtis-TRI You have completely misunderstood me.  GeometrySystem does not have a context.  There is no such member field.  If I want to ask it something, I pass it a Context as a method argument.  (You could imagine that I only hand it some portion of a Context in certain cases, though, for example.)  If I want to ask it about some alternative situation, I pass it a context that denotes that situation.

sherm1, 2018-04-19:
I asked @edrumwri about that. Actually at the moment he only asks for geometric info at the start of the step (which is what's in the current Context). He's not trying to evaluate speculative configurations during the update. Alejandro is trying something fancier.

edrumwri, 2018-04-19:
Responding to Sean: yes, RigidBodyTree "owns" a means for determining
distances, contact points, etc., so it has no such problems.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

SeanCurtis-TRI, 2018-04-19:
@jwnimmer-tri I'm convinced that we're not having the same conversation. Both of us feel we're saying something simple and fundamental that the other is completely missing. For example, in your example, I can't imagine how an MBP has access to a GS context that it can *pass*; it only has access to its own context. So, I'm clearly not understanding what you're suggesting.

SeanCurtis-TRI, 2018-04-19:
I deleted my initial response.

You've suggested variations of both of my initial thoughts -- neither of which I found overly compelling but are both *workable*. I'm going to reason them through a bit further. I *suspect* either path will be considered an abomination in practice but will give them their due diligence nonetheless.

sherm1, 2019-11-08:
I believe this issue is currently moot. Closing.

closed 2019-11-08 by sherm1