RobotLocomotion/drake #8644: pybind: Preserve properties for `pybind11::pybind11` and `pybind11::module` from `pybind11Config.cmake`

EricCousineau-TRI, 2018-04-20:
At present, our usage of `cps2cmake` to provide `pybind11` CMake glue does not preserve all of the properties for `pybind11`'s [namespaced interface library targets](http://pybind11.readthedocs.io/en/stable/compiling.html#advanced-interface-library-target).

We should preserve what upstream incorporates; possibly the simplest solution would be to use `pybind11/tools/pybind11Config.cmake.in`? (Or just re-tune our incorporate via `cps2cmake`)

Relates https://github.com/robotlocomotion/drake-shambhala/pull/98

jamiesnape, 2018-04-20:
We should either use CPS for everything or for nothing at all, so the fix is probably going to be the CPS.

jamiesnape, 2018-04-20:
The question is, do you really want all the properties, or would you like to omit the visibility settings for now?

EricCousineau-TRI, 2018-04-20:
Though it feels dirty, I think that you're right, that we should eliminate the visibility properties from `pybind11::module` pending a functional resolution of #8435.

(Did not see that `pybind11::pybind11` was header-only, while `pybind11::module` and `pybind11::embed` actually linked in `${PYTHON_LIBRARIES}`.)

EricCousineau-TRI, 2018-12-18:
I believe this should be closed as this issue was too general (sorry!) given complications from visibility.
This was mostly resolved by Jamie's PRs, which was used here:
https://github.com/RobotLocomotion/drake-shambhala/pull/98/files#diff-6a69cd1a891f26b7d8e1b7692c8e3d7fR36

The visibility bits are better captured by #8433; added a comment there to check CMake interface too.

NOTE: Marginally relates https://github.com/pybind/pybind11/pull/1373 (for upstream updates)

closed 2018-12-18 by EricCousineau-TRI