RobotLocomotion/drake #8648: Add support for differentiating piecewise expressions

avalenzu, 2018-04-20:
It would be great if `if_then_else(f, e1, e2).Differentiate(x)` could return `if_then_else(f, e1.Differentiate(x), e2.Differentiate(x))`. There are obvious pitfalls due to non-differentiability at the junction of the pieces, but this would be a useful behavior to support.

soonho-tri, 2018-04-20:
Just checked that it's how Matlab's symbolic library is handling `piecewise` expressions:

```matlab
y = piecewise(x < 0, 2*x^2, 0 < x, -3*x^2)

diff(y)
ans = piecewise(x < 0, 4*x, 0 < x, -6*x)
```

sherm1, 2018-04-23:
FWIW:
> How about abs, min, and max? 

:+1: Personally I've found it useful to make those derivatives total functions with some reasonable derivative at the junction (i.e., one side or the other).

In general for partial functions would it be reasonable always to define a hidden `else` case that aborts if encountered at run time? That turns it into a sort-of-total function.

soonho-tri, 2018-07-13:
While working on adding symbolic support to constraints in trajectory optimization (e.g. `DirectCollocationConstraint`), I found that we need the proposed extension to support systems whose dynamics involve conditionals or other non-differentiable functions (such as `min`, `max`). `SimpleCar` class is one example.

I think one thing we can do is to keep the current `Expression::Differentiate` as it is, but add a new method which provides the desired semantics. What do you think? If you think this is a good idea, what would be a good name for this method?

ggould-tri, 2020-12-22:
This issue seems to have stalled out with the proposal to solve it via new method.

I find myself wanting this feature, but I wonder if anyone still cares strongly about the current semantics -- I'm not sure how many people are currently using Differentiate at all, much less relying on it to throw on incomplete derivatives..

avalenzu, 2021-01-04:
I'm definitely in favor of dropping the current "throw on incomplete derivatives" semantics.

ggould-tri, 2021-01-04:
One option would be to paint the shed optional-argument colored -- add an `allow_incomplete=false` defaulted arg to the method.  I am hesitant to add optional args everywhere -- we don't want to end up like scipy, after all -- but if anyone relies on the current behaviour it would let us provide best-effort derivatives as an opt-in feature.

ggould-tri, 2021-01-04:
An interesting thing to note here is that we already handle the other two tricky cases without creation-time exceptions:  Points where neither the x nor dx/dy is defined (e.g. negative sqrt) and points where x is undefined but dx/dy is defined (e.g. the singularities of tan).

This might be an argument that @sherm1's proposal, or a NaN (ugh!) equivalent of it would be the consistent semantics and should be default.

jwnimmer-tri, 2021-01-04:
I think that symbolic's treatment of NaN is already on point here, and exactly what you want.  The differentiated expression would be an if-else chain like:
- if left of discontinuity, return lhs derivative;
- else if right of discontinuity, return rhs derivative;
- else return NaN

When _evaluating_ that expression, if we reach the NaN case (e.g., when x == 0), then we would throw an exception (NaNs always do this, for symbolic).  Otherwise, an _unevaluted_ NaN hidden under an if-else can be constructed without error.

ggould-tri, 2021-01-04:
While I'd stop short of saying that symbolic's treatment of NaN is ever what I want, it's definitely the consistent API surface.  I'll PR those semantics and see what people think.

ggould-tri, 2021-01-04:
(In particular I dislike symbolic's treatment of NaN in that it creates a situation where an Evaluate() call can throw but the caller cannot determine ahead of time if it will do so; combined with the no-catch rule, this amounts to requiring callers to take a leap of faith on each evaluation.  That was defective in the MBP GetFooByName methods and it's likewise defective here, but it's blessed by tradition and documentation.)

jwnimmer-tri, 2021-01-04:
For background: Because our symbolic simplifications are unsound under NaN, we can't ever really handle them well -- we'll always need some concept of a partial function.  It would be a fair feature request to be able to _interrogate_ the expression for whether it will throw under some environment (or whether _there exists_ some environment where it _might_ throw).

For my own taste, I'd still vote to stick with execeptions (and not, e.g., switch to returning a std::optional instead), at least until we have std::expected ready to go.  The common case of no-exception is too awkward with std::optional cropping up.

ggould-tri, 2021-01-04:
Yup.  Unfortunately the much-needed interrogation method (eg, `Expression::is_nan(Environment)`) is difficult to implement without exceptions in the current design (see eg the handling of domain errors), so I don't expect to be able to fix it soon.

ggould-tri, 2021-01-11:
Per the above PR #14514, the location of the discontinuities in if-then-else is a bit wonky.  That PR tosses a `Differentiate`-time exception for non-relational predicate `Formula`s and puts the discontinuity at the equality point of a relational `Formula`.  This is both over-conservative and architecturally incorrect.  A better solution would be to put a `Discontinuities` method into `Formula` itself that can return an empty (constant formulae), single (relational formulae), recursively evaluated (boolean formulae), or something else (the others) collection of discontinuity values.  This is TODO'ed in the PR.  Depending on the complexity of implementation, this might want to be an issue in its own right.

jwnimmer-tri, 2021-11-18:
I found it difficult to understand the current status of the issue, given the above text.  Is the below a fair summary?

In cases where the boundary condition of the piecewise expression is trivial (abs, min, max, ceil, <, >, >=, <=, etc.), then this feature has already been implemented.  At the (single) boundary point, it produces NaN (i.e., any evaluation at that point will throw).

In cases where the boundary condition(s) are more complicated (conjunction, disjunction, boolean variable, quantification, etc.), then the call to `Differentiate` itself will immediately throw, because we have a TODO do add a `Discontinuities` function.

The call to action would be either to implement the `Discontinuities` function so that `Differentiate` will not throw, or to close the issue as "works in sufficiently many cases for now; re-open if anyone needs more".

WDYT @ggould-tri ?

ggould-tri, 2021-11-18:
I believe that you have it right.  To put it directly:

* IF_THEN_ELSE(p, x, y) has an undefined derivative where:
  * (trivially) where the selected x or y has an undefined derivative,
  * (a) where p crosses zero, and
  * (b) where p itself is discontinuous.
* Condition a can be determined in cases of "relational" predicates (equalities, inequalities, and their friends) by current code.  It cannot be determined for non-relational predicates in current code (but these probably are too rare to matter).
* Condition b cannot be determined in current code, or indeed in our current API design, although in principle it frequently can be determined.
* Lots of things are lowered to IF_THEN_ELSE during evaluation or differentiation, so this issue is not limited to explicit conditionals.

And as you say, the advice is "this will probably work for you because you live in condition a; if it doesn't then you need to implement `Discontinuities` to implement condition b".  There are also the non-relational instances of condition a but I think they are silly.

jwnimmer-tri, 2021-11-18:
Okay, thanks!  In that case I will deem this "sufficiently fixed" (with a nice TODO already in place).

If anyone finds that they need to symbolically differentiate a more complicated form of expression that is _not_ covered by the current implementation, please either open another feature request issue, or post back here and we can re-open this one.

closed 2021-11-18 by jwnimmer-tri

ggould-tri, 2021-11-18:
The TODO should be updated to reference this discussion or nobody will ever find this text.

PR #16127 (2021-11-19): [symbolic] Cite issue number in TODO
Closes #8648.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16127)