RobotLocomotion/drake #8711: sphinx docs need a more visible pointer to shambhala

RussTedrake, 2018-04-28:
I think it's time to incorporate shambhala more fully into the drake documentation.   Probably also a visible pointer to http://underactuated.mit.edu as a source of many more examples in the same location?

Perhaps right on the installation and distribution page?

Motivated by answering a lot of questions like this:
https://stackoverflow.com/questions/50060927/how-to-use-robotlocomotion-drake-codes-in-my-project-for-example-qp-inverse/50075307#50075307

jwnimmer-tri, 2018-04-28:
SGTM.

Possibly in concert with a Shambhala top-level README.md addition that says something like:

_This repository shows you how to use Drake from your own project.  Choose one of the three subdirectories named `drake_*` as your guide, depending on which mechanism is most suitable for you._

as a first-paragraph introduction, relegating the metaphysics to a "Where does the name Shamhala  come from?" footnote instead of the first thing a user sees.

jamiesnape, 2018-04-30:
> as a first-paragraph introduction, relegating the metaphysics to a "Where does the name Shamhala come from?" footnote instead of the first thing a user sees.

I would be more controversial and say that we should give the repo a more descriptive (and less exciting name) than "drake-shambhala". Same for the ROS examples repo beginning with "k" (whose full name that I cannot remember offhand) should it ever graduate to RobotLocomotion.

RussTedrake, 2018-09-23:
Want to re-assert the importance of this.  

Proposal:
drake-shambhala => drake-as-external ?  drake-external-project ?  drake-as-a-library ?  drake-in-your-project ? 

https://github.com/naveenoid/kumonoito => RobotLocomotion/drake-ros ?

EricCousineau-TRI, 2019-02-13:
@jamiesnape Given that you've merged this in, how hard would it be to resolve this?

Perhaps we can consolidate these references to just point to them on the front page?

jamiesnape, 2019-02-13:
Merged what? Resolve what? To what this issue is pertaining seems to be different in almost every comment.

EricCousineau-TRI, 2019-02-13:
Sorry, meant the rename. Updated comment.

RussTedrake, 2019-03-20:
Ping on this.  Can we resolve it?

EricCousineau-TRI, 2019-03-21:
This can be done in or after #10963.

commit 7a28469f (2019-03-21): Clean up drake front page

New logo subtitle.
Replace long text list of features with a simple table.
First pass has some simple graphics and links to the c++ doxygen.  I plan to fill in each of these boxes with links to sphynx tutorials (using pydrake) in the next passes.We can also make better images.

Zap some old, stale documentaiton.
Zap outdated models text referenced in #9287
Resolves #8711

Added some follow-on TODOs:
closed 2019-03-21 by jamiesnape