RobotLocomotion/drake #8717: Block network connections for unit tests by default?

jamiesnape, 2018-04-30:
Obviously we would need to open up connections to license servers, but feels like we should make use of the Bazel functionality and declare when we need a network connection, which would also help identify possible sources of flakiness.

jwnimmer-tri, 2019-10-30:
I tried this briefly in Anzu and gave up.  My recollection is that the `block-network` tag makes the `socket()` syscall fail, even if the use was spurious, or even if the only subsequent use was going to be to listen on loopback for a connection that never comes, etc.  My appraisal was that we'd need to whitelist a lot of tests, to the point where it wasn't going to add net value.  The tag doesn't block network _connections_, it blocks _ever calling any networking-related APIs_.

jamiesnape, 2019-10-30:
Let's close then.

closed 2019-10-30 by jamiesnape