RobotLocomotion/drake #8718: Support <include> in the MBP parser

amcastro-tri, 2018-04-30:
We'd like to support the workflow of adding multiple robots and models to the world coming from different files. That is, robot1 in sdf1, robot2 in urdf1, table in urdf2, etc.

With the MBP parser built on top of `sdformat` it'd seem this capability is supported with the `<include>` tag. Other options are viable but this one would seem to be the most useful to users.

We'd like working examples of this once #8687 lands.

cc'ing @avalenzu, @jwnimmer-tri

avalenzu, 2018-04-30:
We should test that the parameters of the `MBP` that results from parsing an SDF that includes a single URDF match those of a `RigidBodyTree` built from the same URDF. That will be an important guarantee to have as we replace programatic assembly of URDFs with SDFs that describe the entire scene.

nkoenig, 2018-04-30:
Parsing the <include> tag should only require telling libsdformat how to find your included SDF files. This can be accomplished through the use of [sdf::setFindCallback](https://bitbucket.org/osrf/sdformat/src/278eefccef6bb42ad7e4c7eb708ef2edc8f3c925/include/sdf/SDFImpl.hh#SDFImpl.hh-73).

I can add other alternatives, such as environment variable based paths.

jwnimmer-tri, 2018-04-30:
For what its worth -- ideally, any configuration that affects parsing (such as the find-path resolution) could be done with a parse-specific context (e.g., an extra argument to the routine that begins a parse), instead of a global variable.  Drake won't necessarily be the only user of sdformat, so we don't want to lay claim to any particular value of its global settings.

sammy-tri, 2018-07-11:
Does this count as a duplicate of #5317 ?  (or should that be closed and new work tracked here?)

sammy-tri, 2018-08-21:
I think now that #9159 has landed we can go ahead and close this issue.  We can open more specific issues for new behavior we want to add/change/remove in the future.  (e.g. I vote for testing/supporting URDF related behavior in another issue)

closed 2018-08-21 by sammy-tri