RobotLocomotion/drake #8719: Investigate a failure on mac machine with rotation constraint

hongkai-dai, 2018-04-30:
In https://github.com/robotlocomotion/drake/blob/cc8ca9df00e49dd79616e4d7d1be052863732a18/solvers/rotation_constraint.cc#L459 I tried to add a "polishing" step on the Mosek optimization result, as
```C++
  // The optimization problem might not be solved to super high precision. To
  // guarantee that the constraints d <= nᵀ * pts.col(i), nᵀ * n = 1 are
  // satisfied, we "polish" the optimization solution below.
  n->normalize();
  *d = std::numeric_limits<double>::infinity();
  for (const auto& pt : pts) {
    *d = std::min(*d, n->dot(pt));
  }
```
But then using this polished result, Gurobi 7.5 fails to find a solution on Mac machine. I should have investigated why Gurobi fails.

jwnimmer-tri, 2023-05-23:
Given 5+ years passed, and a few Gurobi upgrades, perhaps this is no longer worth tracking?  We could close as "won't fix".

hongkai-dai, 2023-05-23:
We can close this issue.

closed 2023-05-23 by hongkai-dai