RobotLocomotion/drake #8750: Support type-erased pixel type in `Image<>`

EricCousineau-TRI, 2018-05-03:
It'd be convenient for writing API's if there were a type-erased version of `Image<>`, which could simply support a closed set (or open set, if we're feeling up to it) of image types, perhaps defining a flag type `PixelType { ... kDynamic }`, use either virtual dispatch or some C++14 variant of `std::any`, and do some API trickery to make it appear similar.

\cc @siyuanfeng-tri @kunimatsu-tri

hongkai-dai, 2019-03-25:
@EricCousineau-TRI is this feature still needed?

EricCousineau-TRI, 2019-03-25:
I would like to keep it open, yes.

jwnimmer-tri, 2023-05-23:
Would adding this typedef to Drake be sufficient to close the issue?

```c++
using ImageVariant = std::variant<ImageBgr8U, ImageBgra8U, ImageDepth16U,
                                  ImageDepth32F, ImageExpr, ImageGrey8U,
                                  ImageLabel16I, ImageRgb8U, ImageRgba8U>;
```

jwnimmer-tri, 2024-03-16:
We have `using ImageAny = std::variant<...>` now.  That might solve the problem.

Since we lack a clearly-articulated use case for something more complicated, we'll close this as "not planned".

closed 2024-03-16 by jwnimmer-tri