RobotLocomotion/drake #8759: misleading error message in piecewise_polynomial.cc

ggould-tri, 2018-05-03:
(scribing this instead of just fixing it because I don't have a drake developer setup here)

If you have points that are too close together in time, piecewise polynomial emits a message that points must be increasing; this is aggravating because you can get this error even when points are increasing.

The epsilon comparison at piecewise_polynomial.cc:367 should be reflected in the error message below, which could even mention the values compared.

Better still would be a justification of why a static epsilon is desirable -- the epsilon is (IMO) many orders of magnitude higher than numeric concerns can justify, and Drake can't really know what timescale its callers are living in.

jwnimmer-tri, 2018-05-07:
@RussTedrake any opinions on the correct fix to the hard-coded epsilon (1e-10) question?

jwnimmer-tri, 2019-10-10:
@ggould-tri Perhaps you'd like to take a whack at fixing this now?

PR #12183 (2019-10-14): #8759:  Correct misleading error message / documentation
Closes #8759.

This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12183)
commit 54aa4a47 (2019-10-14): #8759:  Correct misleading error message / documentation (#12183)

* Fix #8759:  Correct misleading error message / documentation
closed 2019-10-14 by ggould-tri