RobotLocomotion/drake #8761: SDFormat Errors when running executable directly

avalenzu, 2018-05-04:
```
$ bazel test //multibody/multibody_tree/parsing:multibody_plant_sdf_parser_test
```
runs just fine. However, if I then run
```
$ bazel-bin/multibody/multibody_tree/parsing/multibody_plant_sdf_parser_test
```
I get
```
Using drake_cc_googletest_main.cc

[==========] Running 2 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 2 tests from AcrobotModelTests
[ RUN      ] AcrobotModelTests.ModelBasics
Error [parser.cc:77] Unable to find or open SDF file[root.sdf]
unknown file: Failure
C++ exception with description "From AddModelFromSdfString():
Error: SDF does not have a version.
" thrown in SetUp().
[  FAILED  ] AcrobotModelTests.ModelBasics (1 ms)
[ RUN      ] AcrobotModelTests.VerifyMassMatrixAgainstBenchmark
Error [parser.cc:77] Unable to find or open SDF file[root.sdf]
unknown file: Failure
C++ exception with description "From AddModelFromSdfString():
Error: SDF does not have a version.
" thrown in SetUp().
[  FAILED  ] AcrobotModelTests.VerifyMassMatrixAgainstBenchmark (1 ms)
[----------] 2 tests from AcrobotModelTests (2 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test case ran. (2 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 2 tests, listed below:
[  FAILED  ] AcrobotModelTests.ModelBasics
[  FAILED  ] AcrobotModelTests.VerifyMassMatrixAgainstBenchmark

 2 FAILED TESTS
```
Perhaps it's an issue with finding `root.sdf` in the runfiles? cc. @nkoenig, @jwnimmer-tri

avalenzu, 2018-05-04:
I've compared the behavior in the two cases by adding printouts to `sdf::findFile()` in `SDF.cc`. It looks like `root.sdf` is found in the "Now check the install path" portion of that function when run under Bazel, but not when run directly. 

Digging a bit further, this appears to be because `SDF_SHARE_PATH` is set to `external/sdformat/sdf/1.6/` when `sdf_config.h` is generated from `sdf_config.h.in`. It looks like that path exists when running under Bazel, but not when running the executable directly? Not sure what should be done about that ...

amcastro-tri, 2018-05-04:
I believe this is the same as issue #7874? 
cc'ing @nkoenig, @jamiesnape.

amcastro-tri, 2018-06-12:
A fix was merged into sdformat, [#434](https://bitbucket.org/osrf/sdformat/pull-requests/434/embed-sdf/diff).
I believe for this to work for us we need some python magic? @EricCousineau-TRI?

EricCousineau-TRI, 2018-06-12:
Yup, this is the Python version script, at least at the revision that I had seen:
https://github.com/EricCousineau-TRI/sdformat/commit/4bad9c20555a7350e46f629e7a0fb5222a4260f1#diff-358fe4d439e264bddb6b1a37d6ca8d54

I'm off in OSPRay land trying to debug stuff 'n such, so I may not be able to get to it super quick.
What time you need it by? Might Friday be OK?

amcastro-tri, 2018-06-12:
That'd work. Thanks!

jwnimmer-tri, 2018-06-12:
Can we close this as a dup of #7874 so that we only have one place to trace progress updates and assignments?

EricCousineau-TRI, 2018-06-12:
Duplicate of #7874

closed 2018-06-12 by EricCousineau-TRI