RobotLocomotion/drake #8764: Unit tests rgbd_renderer_vtk_test and rgbd_camera_test Fail on Ubuntu 16.04

liangfok, 2018-05-06:
I'm encountering the following problems on Ubuntu 16.04 even after running `sudo ./setup/ubuntu/16.04/install_prereqs.sh`:

```
$ bazel test ...
Starting local Bazel server and connecting to it...
..........
INFO: Analysed 5985 targets (307 packages loaded).
INFO: Found 2488 targets and 3497 test targets...
FAIL: //systems/sensors:rgbd_renderer_vtk_test (see /home/liang/.cache/bazel/_bazel_liang/b4934af35f1751eb80b7d0879c04e193/execroot/drake/bazel-out/k8-opt/testlogs/systems/sensors/rgbd_renderer_vtk_test/test.log)
INFO: From Testing //systems/sensors:rgbd_renderer_vtk_test:
==================== Test output for //systems/sensors:rgbd_renderer_vtk_test:
Using drake_cc_googletest_main.cc

[==========] Running 8 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 8 tests from RgbdRendererVTKTest
[ RUN      ] RgbdRendererVTKTest.InstantiationTest
[       OK ] RgbdRendererVTKTest.InstantiationTest (91 ms)
[ RUN      ] RgbdRendererVTKTest.NoBodyTest
[       OK ] RgbdRendererVTKTest.NoBodyTest (78 ms)
[ RUN      ] RgbdRendererVTKTest.TerrainTest
================================================================================
FAIL: //systems/sensors:rgbd_camera_test (see /home/liang/.cache/bazel/_bazel_liang/b4934af35f1751eb80b7d0879c04e193/execroot/drake/bazel-out/k8-opt/testlogs/systems/sensors/rgbd_camera_test/test.log)
INFO: From Testing //systems/sensors:rgbd_camera_test:
==================== Test output for //systems/sensors:rgbd_camera_test:
Using drake_cc_googletest_main.cc

[==========] Running 7 tests from 3 test cases.
[----------] Global test environment set-up.
[----------] 1 test from RgbdCamera
[ RUN      ] RgbdCamera.TestInstantiation
[       OK ] RgbdCamera.TestInstantiation (176 ms)
[----------] 1 test from RgbdCamera (176 ms total)

[----------] 2 tests from RgbdCameraDiagramTest
[ RUN      ] RgbdCameraDiagramTest.FixedCameraOutputTest
================================================================================
INFO: Elapsed time: 52.133s, Critical Path: 3.69s
INFO: 3 processes, local.
INFO: Build completed, 2 tests FAILED, 4 total actions
//systems/sensors:rgbd_camera_test                                       FAILED in 0.6s
  /home/liang/.cache/bazel/_bazel_liang/b4934af35f1751eb80b7d0879c04e193/execroot/drake/bazel-out/k8-opt/testlogs/systems/sensors/rgbd_camera_test/test.log
//systems/sensors:rgbd_renderer_vtk_test                                 FAILED in 0.6s
  /home/liang/.cache/bazel/_bazel_liang/b4934af35f1751eb80b7d0879c04e193/execroot/drake/bazel-out/k8-opt/testlogs/systems/sensors/rgbd_renderer_vtk_test/test.log

Executed 3 out of 3497 tests: 3495 tests pass and 2 fail locally.
There were tests whose specified size is too big. Use the --test_verbose_timeout_warnings command line option to see which ones these are.
```

I tried both suggestions mentioned in the FAQ (i.e., setting "local = 1" in the BUILD file and using "--spawn_strategy=standalone") to no avail.

liangfok, 2018-05-06:
/cc: @stonier and @rpoyner-tri because I think we discussed this earlier this week.

jwnimmer-tri, 2018-05-07:
I think this is a duplicate of #7908.  It is a known issue that, with certain NVidia configurations, the tests fail.  Does turning on show_window = true cause the test to pass for you?

EricCousineau-TRI, 2018-05-11:
If you are running `bazel test //...`, you can change it system-wide via the flag added here:
https://github.com/RobotLocomotion/drake/pull/8006

PR #8789 (2018-05-16): Update to VTK 8.1 and latest Director and add packages for Bionic
Needs local testing on Linux of `drake-visualizer` using both `bazel run //tools:drake_visualizer` and the installed executable (`bazel run install -- /opt/drake && /opt/drake/bin/drake-visualizer`). Note that for licensing reasons this switches to using a shared library for LCM (c.f., #7423, #8153).

Fixes #7908, fixes #8322, fixes #8764. Relates #7907.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8789)