RobotLocomotion/drake #8795: LCM Java Errors on build

wsmoses, 2018-05-11:
I'm getting the following LCM java-related errors when I try to build from source on ubuntu.

Analyzing: 6083 targets (0 packages loaded)
INFO: Analysed 6083 targets (0 packages loaded).
INFO: Found 6083 targets...
[4 / 35] [-----] Expanding template bindings/pydrake/systems/py/drawing_py_drakelint
ERROR: /home/wmoses/.cache/bazel/_bazel_wmoses/93a6e4db8fc6f38a669104ff64569d2a/external/optitrack_driver/lcmtypes/BUILD.bazel:33:1: Building external/optitrack_driver/lcmtypes/liblcmtypes_optitrack.jar (7 source files) failed (Exit 1)
bazel-out/k8-opt/genfiles/external/optitrack_driver/lcmtypes/optitrack/optitrack_data_descriptions_t.java:12: error: [strict] Using type lcm.lcm.LCMEncodable from an indirect dependency (TOOL_INFO: "@lcm//:lcm-java"). See command below **
public final class optitrack_data_descriptions_t implements lcm.lcm.LCMEncodable
                                                                   ^
bazel-out/k8-opt/genfiles/external/optitrack_driver/lcmtypes/optitrack/optitrack_data_descriptions_t.java:27: error: [strict] Using type java.util.ArrayList from an indirect dependency (TOOL_INFO: "/home/wmoses/.cache/bazel/_bazel_wmoses/93a6e4db8fc6f38a669104ff64569d2a/execroot/drake/bazel-out/host/genfiles/external/bazel_tools/tools/jdk/platformclasspath.jar").
        LCM_FINGERPRINT = _hashRecursive(new ArrayList<Class<?>>());
                                             ^
 ** Please add the following dependencies: 
  @lcm//:lcm-java  to @optitrack_driver//lcmtypes:lcmtypes_optitrack 
 ** You can use the following buildozer command: 
buildozer 'add deps @lcm//:lcm-java ' @optitrack_driver//lcmtypes:lcmtypes_optitrack 

[6,521 / 8,287] Building external/lcmtypes_bot2_core/liblcmtypes_bot2_core_java.jar (38 source files); 0s worker ... (2 actions running)
INFO: Elapsed time: 4.886s, Critical Path: 1.56s
FAILED: Build did NOT complete successfully

EricCousineau-TRI, 2018-05-14:
Just to check, can I ask which version of Ubuntu you're using, and if you were able to run `install_prereqs` without any errors?

\cc @jamiesnape Can I ask if you've seen this before?

jamiesnape, 2018-05-14:
No, I have not. Which version of Bazel and which SHA of Drake?

jwnimmer-tri, 2018-09-27:
Closing for lack of info.  See also https://drake.mit.edu/getting_help.html.

closed 2018-09-27 by jwnimmer-tri