RobotLocomotion/drake #8798: Conflict between default_scalars and system_scalar_converter

SeanCurtis-TRI, 2018-05-11:
In a recent PR (#8778) a confusing compiler error arose.

Problem Statement
==============
The `QuadrotorPlant` uses the `SystemScalarConverter` to get its transmogrification for free. In `quadrotor_plant.cc` it uses `default_scalars` to instantiate on *nonsymbolic* scalars. Prior to the PR in question, both of those features were fine.

In the PR, some functionality was changed such that some of the underlying components of `QuadrotorPlant` no longer supported the `symbolic::Expression` scalar. (In this case, it's the new `RotationMatrix` class.)

The compiler warned that it couldn't instantiate a `RotationMatrix<Expression>` type (which is fair because such a thing does not exist.) However, the author of the PR was mystified why this error was even coming up. The `QuadrotorPlant` was only being instantiated on `double` so it wasn't obvious why the compiler was even looking for `Expression`-valued types.

THe solution, of course, was to specialize the conversion traits for the plant in question to be `NonSymbolicTraits`. Now compilation proceeded apace.

__It seems odd that the conversion table can be set up to include instantiations of a class that don't actually exist (in this case the `QuadrotorPlant<Expression>`).__

Solution(?)
======

__Option 1:__ It would be *great* if it wasn't possible to convert to an unsupported scalar type. I.e., if I only instantiate it on numeric scalar types, it will only convert to those same supported types. This would happen with a single declaration without my having to coordinate two different calls.

__Option 2:__ An error message that can better direct users who are less versed in the ways of transmogrifcation would mitigate the personal cost of otherwise addressing inscrutable errors.

jwnimmer-tri, 2018-05-11:
> ... class that don't actually exist ...

But it does.  The `DRAKE_DEFINE_CLASS_TEMPLATE_INSTANTIATIONS_ON_DEFAULT_SCALARS` just makes _at least_ those types exist (to ensure that they compile in their entirety).  But since the class in question is templated, other instantiations will come into being as-needed, when something refers to them (and has the method definitions etc. available) -- just like you can use `std::vector<MyThing>` without the author of `vector` knowing that `MyThing` is a thing.

I don't immediately see a way to do either option 1 or 2, but I'll give it more thought.

SeanCurtis-TRI, 2018-05-11:
And on the subject of:

>  other instantiations will come into being as-needed when something refers to them.

The `QuadrotorPlant` class is one of those with the meat of its implementation in the .cc file. Meaning that if it isn't part of the default scalar declaration, it *can't* exist.  Given that one of those happens in the .cc file and the other happens in the header file, I don't see any way to coordinate those. 

But I have full faith in your ingenuity and experience to find us a better way.

jwnimmer-tri, 2018-05-11:
I think the best we'll be able to do is to improve the documentation in `system_scalar_conversion_doxygen.h` to insist that when using SystemTypeTag the traits specialization (or lack of) and the `DRAKE_DEFINE_...` are congruent, and then make sure the platform reviewers know to check that during code review.

jwnimmer-tri, 2018-05-11:
I guess the other options is to add sugar for `DRAKE_DEFINE_CLASS_TEMPLATE_INSTANTIATIONS_ON_DEFAULT_NONSYMBOLIC_SCALARS` that _also_ sets the traits, and have System subclasses use the sugar instead.  But that would only help when enough code is in the cc file, which is a bit more rare.  I think starting with documentation is best.

We could also audit the current master against that proposed consistency rule, and see if there are other lurking surprises.

SeanCurtis-TRI, 2018-05-11:
Well, if that's the best we can do, that's better than now. Thanks

SeanCurtis-TRI, 2018-05-11:
quadrotor_plant will be corrected in the referenced PR.

SeanCurtis-TRI, 2018-05-17:
Quick concern about documentation:  in [`scalar_conversion_traits.h`](https://github.com/RobotLocomotion/drake/blob/master/systems/framework/scalar_conversion_traits.h#L25) we have the following documentation:

```
/// In unsupported cases, the constructor will not even be mentioned by the
/// framework, so that S need not even compile for certain values of T and U.
```

This statement is strictly true. However, it can be misleading without careful reading. The key word is "supported". It's implicitly defined earlier in the docs as:

> Particular scalar-dependent classes may specialize this template to indicate whether the framework should __support__ conversion for a combination of S, T, and U.

However, the concept of "supported" is overloaded and I worry that the default interpretation is not what we intend.  Really what this is all about is *declaration*. This implicitly declares what is supported and if there is conflict between this declaration and what is implemented, then we get the conflict observed in @mitiguy's PR.

How do you feel about including this in a documentation tweak? Something along the lines of this being a declaration of supported conversion functionality and it is important for the declared conversions to be supported by `S`.

PR #8833 (2018-05-21): Tidy up some scalar conversion declarations and documentation.
See commit messages for details.

Closes #8798.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8833)
closed 2018-05-21 by jwnimmer-tri