RobotLocomotion/drake #8820: Need method that provides fractional length of maliput::api::Lane

liangfok, 2018-05-16:
Currently, `maliput::api::Lane::length()` returns the entire length of the lane from `s = 0` to `s = s_max`. I need a fractional length of the lane in the range `(0, s_max)`. The situation is as follows:

I obtained a `LaneSRoute` and want to do a sanity check to verify that its length is approximately equal to the expected length and not some crazy high value. Doing this requires obtaining the lengths of the first and last lanes in the route, which are fractions of the lanes. Thus, I need to have a method like `double maliput::api::Lane::length(int start_s, int end_s) const;`.

liangfok, 2018-05-16:
/cc: @maddog-tri thoughts?

liangfok, 2018-05-16:
Upon further inspection, I now believe the fractional arc-length is simply equal to |s1 - s0|. I shall thus close this ticket.

closed 2018-05-16 by liangfok

maddog-tri, 2018-05-17:
Bingo.

(Corollary:  `s_max` is exactly `maliput::api::Lane::length()`.)