RobotLocomotion/drake #8821: Python unittests should write an XML_OUTPUT_FILE

jamiesnape, 2018-05-16:
We are currently getting very rudimentary output into `test.xml` from the default test runner provided by Bazel. Instead, it would be nice to do something like

```python
import os
import unittest
import xmlrunner

if __name__ == '__main__':
    with open(os.environ['XML_OUTPUT_FILE'], 'wb') as output:
        unittest.main(testRunner=xmlrunner.XMLTestRunner(output=output))
```

(or use a different test runner, there are lots of choices, Bazel for its own testing uses a shell script for whatever reason) to get a similar level of detail to the C++ GTests. This would be helpful for the next generation of Jenkins jobs.

EricCousineau-TRI, 2018-05-16:
I think the solution is simply changing this line of code to what you have:
https://github.com/RobotLocomotion/drake/blob/398a59e7603cea532d7d9ad863d7fb0d1492f776/common/test_utilities/drake_py_unittest_main.py#L102-L103

Given that you have the solution and the best bearing for debugging this, could I reassign this issue back to you?
I can still do it, but there may be additional turnaround time.

jamiesnape, 2018-05-16:
Sure, I can do it.

jamiesnape, 2018-05-16:
(BTW for anyone interested, this is why you get better output for tests that use GTest, https://github.com/google/googletest/blob/045e7f9ee4f969ac1a3fe428f79c4b880f0aff43/googletest/src/gtest-port.cc#L1247-L1262)

jamiesnape, 2018-05-16:
For some reason this is causing `math_overloads_test` to fail catastrophically by trying to open the same file five times over. Anything unusual about that test?

jwnimmer-tri, 2018-05-16:
Yes -- it re-execs itself, it seems: https://github.com/RobotLocomotion/drake/blob/fdac5ba7a64ee773197b29f6462a1391b3230e56/bindings/pydrake/test/math_overloads_test.py#L195-L208

jamiesnape, 2018-05-16:
Wow. Not sure the authors of the test runner (or myself) considered that possibility.

jwnimmer-tri, 2018-05-18:
FWIW I think it would also be easy to have the `BUILD.bazel` rules run the test multiple times, each under different label?  Basically, manually sharding it.  (Or possibly even Bazel's built-in sharding would be good enough?)

jamiesnape, 2018-05-18:
> FWIW I think it would also be easy to have the BUILD.bazel rules run the test multiple times, each under different label? Basically, manually sharding it. (Or possibly even Bazel's built-in sharding would be good enough?)

Personally I think that would be preferable.

jwnimmer-tri, 2019-03-08:
Given the increasing number of python-driven test cases (including in Anzu), I think it would be nice if we could add this feature, even if it means blacklisting the math_overloads_test from xml reporting, for now.

@EricCousineau-TRI Can I ask for a priority bump here?

EricCousineau-TRI, 2019-03-08:
Bumping priority (feel free to make it higher).

W.r.t. `math_overloads_test`, I can certainly de-stupify the test and see about using sharding (or something else that's less dumb).

@jamiesnape Do you think this might be doable with the next week or two?

jamiesnape, 2019-03-08:
If it is high priority, maybe within the next two weeks, otherwise I have a huge list of medium priority items with which it would need to fight for time.

jwnimmer-tri, 2019-03-08:
@EricCousineau-TRI if you want to fix math_overloads_test, I could probably take on the xmlrunner change.  I think I might even just prototype it inside Anzu in the next week, anyway.

jwnimmer-tri, 2020-12-10:
Per #11003, the `math_overloads_test` blocker is gone now.

PR #14560 (2021-01-25): tools: Add xmlrunner and use it for Python unittests
Remove a stray print statement whose unicode defeats the xml runner's output scraping.

Closes #8821.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14560)
closed 2021-01-25 by SeanCurtis-TRI