RobotLocomotion/drake #8831: Error in running prereqs.sh

junhyeokahn, 2018-05-18:
Hello, I am using Mac OS and having some trouble installing drake from source.
When I run the prereqs.sh I got below error.

```
junhyeok ~/Repository/drake ./setup/mac/install_prereqs.sh
+ source ./setup/mac/binary_distribution/install_prereqs.sh
++ set -euxo pipefail
++ [[ 501 -eq 0 ]]
++ command -v /usr/local/bin/brew
++ /usr/local/bin/brew update
Already up-to-date.
++ /usr/local/bin/brew bundle --file=./setup/mac/binary_distribution/Brewfile
Using caskroom/versions
Using dreal/dreal
Using robotlocomotion/director
Using boost
Using cmake
==> Installing dreal from dreal/dreal
Error: Operation already in progress for dreal.formula
Another active Homebrew process is already using dreal.formula.
Please wait for it to finish or terminate it to continue.
Installing dreal has failed!
Error: Operation already in progress for sphinx-doc.formula
Another active Homebrew process is already using sphinx-doc.formula.
Please wait for it to finish or terminate it to continue.
Installing gflags has failed!
Error: Operation already in progress for sphinx-doc.formula
Another active Homebrew process is already using sphinx-doc.formula.
Please wait for it to finish or terminate it to continue.
==> Upgrading 1 outdated package, with result:
glew 2.0.0 -> 2.1.0
Installing glew has failed!
Error: Operation already in progress for pkg-config.formula
Another active Homebrew process is already using pkg-config.formula.
Please wait for it to finish or terminate it to continue.
==> Upgrading 1 outdated package, with result:
glib 2.54.2 -> 2.56.1
Installing glib has failed!
Error: Formulae found in multiple taps:
       * dartsim/dart/ipopt
       * robotlocomotion/director/ipopt

Please use the fully-qualified name e.g. dartsim/dart/ipopt to refer the formula.
```

CatchMeFastFat, 2018-05-18:
see if that helps [Another active Homebrew process is already in progress](https://stackoverflow.com/questions/39797078/another-active-homebrew-process-is-already-in-progress?utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa)

junhyeokahn, 2018-05-18:
Thanks, it helps :)

closed 2018-05-19 by junhyeokahn