RobotLocomotion/drake #8836: snopt library

junhyeokahn, 2018-05-18:
Hi, thank you for your effort maintaining the repository.

I would like to use snopt.
I have a license of it and a lib (.so files).
However, the way drake explicitly set the path for snopt is
```
export SNOPT_PATH=/home/username/Downloads/snopt7.6.tar.gz
```
What is the best way to use snopt in the case of having .so files?
Or could I obtain access to the private RobotLocomotion/snopt GitHub repository?

jwnimmer-tri, 2018-05-22:
I can't speak to the RobotLocomotion repository permission question, but I can speak to using binary snopt.

To be honest, this is the first I've heard of `snopt` being available in binary form.  Previously, I was aware only of two kinds of source releases -- one (source) release for licensed users which has no size limits, and then one (source) release for students that has some problem-size limits built in (see #245).

Does the `so` you have come with headers, or (if you know?) does it assume that the https://github.com/snopt/snopt-interface/tree/master/include headers are the relevant ones?  Is there a way that Drake developer(s) could obtain the same or similar snopt binaries, if we wanted to test our compatibility with it?

Its likely that to use an `*.so` release of snopt, we would have to edit the files in https://github.com/RobotLocomotion/drake/tree/master/tools/workspace/snopt to handle that case, doing a `cc_library()` call with `srcs = ["snopt.so"]` etc.  Take note also of #8825 where those build files are already planned for rework a little bit.

junhyeokahn, 2018-05-22:
Thank you for your kind reply.

> Does the so you have come with headers, or (if you know?) does it assume that the https://github.com/snopt/snopt-interface/tree/master/include headers are the relevant ones? Is there a way that [...]

I have ```so``` with https://github.com/snopt/snopt-interface/tree/master/include headers.

Do you guys have any plan to handle this case by editing build.bazel? Or I could start looking at it in my local computer.

jwnimmer-tri, 2018-05-22:
Issue #7984 tracks porting Drake to use those signatures, instead of the legacy Fortran signatures that we are currently using.  That would be a prerequisite of adding support for a precompiled `so`.  (Once 7984 is solved, the `package.BUILD.bazel` changes to use it would be relatively easy -- change `srcs = []` to be the `*.so` file, instead of the C sources.)

I don't think anyone is actively working on #7984 yet, but that issue would be the place to discuss getting it done.

jwnimmer-tri, 2018-10-28:
As noted above, this will be an easy tweak, once #7984 is solved.  Over to @EricCousineau-TRI for the kitware backlog.

jwnimmer-tri, 2019-09-19:
I think I was wrong that this would be easy.  Drake is stuck using our own build of a patched SNOPT 7.4, because of functionality regressions in 7.6.  I'm sorry to say I don't think we can support `libsnopt.so` runtime loading in the near future.

Perhaps the https://github.com/RobotLocomotion/drake/releases are good enough for some users -- they contain SNOPT code already embedded.

closed 2019-09-19 by jwnimmer-tri

jwnimmer-tri, 2023-02-27:
This came up again [here on stackoverflow](https://stackoverflow.com/questions/75586156/building-drake-with-trial-version-snopt).