RobotLocomotion/drake #8850: RgbdRenderer, RgbdCamera: Expose setting intrinsics matrix; ensure that rendering is accurate to arbitrary intrinsics

EricCousineau-TRI, 2018-05-22:
Per f2f

\cc @naveenoid @siyuanfeng-tri @kunimatsu-tri

kunimatsu-tri, 2018-05-23:
The current `RgbdCamera` and/or `RgbdRenderer` APIs are coming from those of VTK where what we can set is `fov_y`, `width` and `height`. There is no way to set `fov_x` and that means we don't have full control for setting both `fx` and `fy` intrinsic parameters independently.

Here is the code where `fov_y` is set:
https://github.com/RobotLocomotion/drake/blob/master/systems/sensors/rgbd_renderer_vtk.cc#L383

@jamiesnape Can you confirm this comment and point out if there is any way to enable us to set all the camera intrinsic parameters using VTK?

SeanCurtis-TRI, 2018-05-23:
We might consider constructing our own projection matrix and setting the matrix rather than using the syntactic sugar that VTK provides.

jamiesnape, 2018-05-23:
You beat me to it. I think the method is something like `SetExplicitProjectionTransformMatrix(vtkMatrix4x4)`

kunimatsu-tri, 2018-05-23:
Great! Then, can we just ignore `SetViewAngle` or do we need some tricks to make it work?

jamiesnape, 2018-05-23:
You can ignore it, `ComputeProjectionTransform()` is short-circuited before it uses it:

https://github.com/Kitware/VTK/blob/4656844ec61ff83c1438f03c7941126ddc53d8c3/Rendering/Core/vtkCamera.cxx#L1046-L1052

kunimatsu-tri, 2018-05-23:
Cool. Thanks for the pointer!

EricCousineau-TRI, 2018-05-23:
Sorry, I'm a little slow on the uptake; just want to confirm some deets / ask some slightly redundant questions about the  coordinates / units:

The documentation for [`GetProjectionTransformMatrix`](https://www.vtk.org/doc/nightly/html/classvtkCamera.html#a4d9a509bf60f1555a70ecdee758c2753) states:

> Return the projection transform matrix, which converts from camera coordinates to viewport coordinates. [...] The viewport coordinates of a point located inside the frustum are in the range ([-1,+1],[-1,+1],[nearz,farz]).

(1)(a) Let's call the camera coordinates `[x, y, z]` and the viewport coordinates `[U, V, W]`, and the projection+transform matrix `PT`; are the relations per the docs simply this?
`[U, V, W; 1] = PT * [x; y; z; 1]`
Or is the an additional scaling term on the left side?

(1)(b) If (1)(a) holds, then the normalized pixel coordinates are simply `u = U / W`, `v = V / W`, right?

(2)(a) It seems that the documentation seems like it *may* be a tad ambiguous, in that it could imply that the frustrum is defined by the above range, but instead should state that the frustrum is a subset of the defined range (e.g. `[-1, 1, farz]` is in the frustrum, but `[-1, 1, nearz]` is not).
Is this the case?

(2)(b) Does this mean that `U, V` are scaled such that, at `farz`, `[-1, 1]` defines the extent of the image? (From looking at `vtkPerspectiveTransform`, seems so?)

EricCousineau-TRI, 2019-01-31:
Reassigning to @SeanCurtis-TRI per Slack discussion.

hongkai-dai, 2019-03-25:
@SeanCurtis-TRI could you update the status of this issue? Can we close it?

EricCousineau-TRI, 2019-03-25:
> Can we close it?

I'd like to keep it open!

SeanCurtis-TRI, 2019-03-25:
Still an open feature request.

EricCousineau-TRI, 2019-07-31:
I'd move to close this in lieu of #11880. That cool with you?

SeanCurtis-TRI, 2020-05-14:
Missed the question. I agree with closing this.

closed 2020-05-14 by SeanCurtis-TRI