RobotLocomotion/drake #8869: Set SNOPT_PATH environment for CI

jwnimmer-tri, 2018-05-24:
I would like to have `drake-ci` test SNOPT use akin to what users would do.  This relates to #7240, #8031, #8801, #8825.  Initial testing can happen via #8825, I think.

My proposal would be the following:
1. bazel and cmake oss builds:
    - environment variable `SNOPT_PATH` is unset (as currently)
    - cmake configs at their defaults (as currently)
2. bazel and cmake `-snopt` builds:
    - environment variable `SNOPT_PATH=/path/to/somezip` (new)
    - bazel `--config snopt` (as currently)
    - cmake configs `WITH_SNOPT` set ON, but otherwise at their defaults (as currently)
3. bazel `-everything` builds:
    - environment variable `SNOPT_PATH=git` (new)
    - bazel `--config everything` (as currently)
4. cmake `-everything` builds:
    - environment variable `SNOPT_PATH` is unset (as currently)
    - cmake configs `WITH_SNOPT` set ON (as currently) and `WITH_ROBOTLOCOMOTION_SNOPT` set ON (new), but otherwise at their defaults

This proposes using git for `-everything` and zip for `-snopt`, as a way to control the permutation matrix.  I am open to other suggestions.

jwnimmer-tri, 2018-05-24:
For clarification, my plan was to use this issue to agree on the end result, and then for #8825 to merge at least to have the non-zip changes pushed to CI sooner rather than later.  (I'm okay either way if the zip changes come along for the ride, or wait.)  If we're skipping the zip support for now, the configurations that would use zip could just use the robotlocomotion git settings for now.

jamiesnape, 2018-08-13:
For now `SNOPT_PATH` is always set to `git` (https://github.com/RobotLocomotion/drake-ci/commit/00389a3a91cba5e28729068690a4a3c9d7de6daf). I am in the process of refactoring our jobs, so the variation in download type will follow as that happens.

jwnimmer-tri, 2018-08-16:
Sounds good, thanks.

EricCousineau-TRI, 2018-12-03:
Closing, as this seems like it's already been handled.

jwnimmer-tri, 2018-12-03:
To my knowledge, we don't have any CI builds that set SNOPT_PATH=/path/to/archive.tgz (or zip).  We should leave some issue open (or file a new issue) until we have Jenkins coverage for that.

Probably this is the best issue to leave open, though I'm okay if there is another one that I've missed.

jwnimmer-tri, 2019-06-20:
Given the low (zero?) rate of bugs we're seeing in `tools/workspace/snopt/repository.bzl` related to path handling, I think this can be Priority: Low now.

jwnimmer-tri, 2020-05-14:
... and for the same reason, we can close it until we start having bugs slip through CI.

closed 2020-05-14 by jwnimmer-tri