RobotLocomotion/drake #8899: Multilane Technical debt - Missing tests

agalbachicar, 2018-05-30:
The following TODOs in code are there to remember addressing tests / modifying tests:

- [ ] [multilane_builder_test.cc:23](https://github.com/RobotLocomotion/drake/blob/master/automotive/maliput/multilane/test/multilane_builder_test.cc#L23) --> Trivial.
- [ ] [multilane_loader_test.cc:42](https://github.com/RobotLocomotion/drake/blob/master/automotive/maliput/multilane/test/multilane_loader_test.cc#L42) and [multilane_loader_test.cc:45](https://github.com/RobotLocomotion/drake/blob/master/automotive/maliput/multilane/test/multilane_loader_test.cc#L45) --> Trivial.
- [ ] [multilane_loader_test.cc:549](https://github.com/RobotLocomotion/drake/blob/master/automotive/maliput/multilane/test/multilane_loader_test.cc#L538) --> implies creating a Group interface and then refactoring that test.

hidmic, 2018-08-07:
Taking this one.

agalbachicar, 2018-08-14:
Taking the third bullet.

agalbachicar, 2019-06-10:
Closing this issue as it has already been migrated to downstream repository.

closed 2019-06-10 by agalbachicar