RobotLocomotion/drake #8904: For APIs that set values, disprefer unique_ptr and prefer const-ref

jwnimmer-tri, 2018-05-30:
From discussion in #8862, we want a meta-issue listing all of the places that need similar treatment within the framework.  I surveyed current master for public methods that take unique_ptr arguments, and excluded APIs that are for internal use only.  This issue captures APIs that should probably switch to use const-ref.

The APIs still to be considered for rework are:
- [ ] ContextBase::FixInputPort (for AbstractValue)
- [ ] Context::FixInputPort (for BasicVector) (per #8862)
- [ ] LeafSystem::DeclareAbstractState
- [ ] Parameters::set_abstract_parameters
- [ ] Parameters::set_numeric_parameters
- [ ] System::DeclareAbstractState

\CC @EricCousineau-TRI FYI

sherm1, 2018-05-31:
This still leaves other ways users can err with cached values. Any method that returns a mutable reference to a raw quantity, for example `ContinuousState::operator[]` or `ContinuousState::get_mutable_vector()` currently permit unchecked modification. More subtly, even methods returning _const_ references introduce the possibility that a held reference becomes invalid. Even more subtly, that can be very useful behavior in the right hands, saving expensive copies during tricky multi-stage updates such as occur during integration steps.

My thought is that we should migrate away from anything allowing unchecked Context modification, including the unique_ptrs mentioned above, and any mutable references, but leave the const references with suitable warnings.

rpoyner-tri, 2025-03-18:
As of #22738, `pydrake` can no longer use unique_ptr arguments, and has very limited use of unique_ptr even in return values. I will claim that this resolves the primary problem of this issue. If @sherm1 wants to file a new issue about cache invalidation subtleties, that's up to him.

closed 2025-03-18 by rpoyner-tri