RobotLocomotion/drake #8908: `//examples/kuka_iiwa_arm:kuka_simulation` violates ODR due to `:iiwa_common` and `:iiwa_lcm`

EricCousineau-TRI, 2018-05-31:
At present, `iiwa_common` and `iiwa_lcm` depend on static libraries within Drake (e.g. `//multibody:rigid_body_tree`), but `kuka_simulation` also links against `libdrake.so` via `drake_example_cc_binary`.

In drafting #8906, I realized this when digging into an issue captured by the following comments:
https://github.com/EricCousineau-TRI/drake/blob/59d9d37d8d6a5fb427b5872a8d85ed03698e756e/tools/skylark/drake_cc.bzl#L611-L617

With @sammy-tri's help, we found that when the example binary has `srcs = ["//tools/install/libdrake:libdrake.so", ...]`, Bazel or the linker somehow magically realize that there are some duplications in the object code, and culls duplicate code from the resultant binary `kuka_simulation`.

However, when moving `libdrake.so` to be in `drake_shared_library`'s `srcs`, it seems that Bazel or the linker do not catch this case, and ends up duplicating globals such as `std::set<int> RigidBodyTreeConstants::default_model_instance_id_set` (which is a GSG violation, as it is not trivially destructible).
This manifests itself when the binary is running its exit handlers, and it ends up double-freeing memory within `default_model_instance_id_set`.

Potential solutions:
- Avoid ODR violations, and somehow ensure that `iiwa_common` and `iiwa_lcm` do not double-declare their dependencies.
- Change `default_model_instance_id_set` to `never_destroyed<>` as a stop-gap. This should be done regardless, but won't solve the issue since there will be other violations.

Ultimately, this seems to be pointing towards the pains of trying to off both static and dynamically linked libraries from Bazel without a concrete plan of offering both types of targets. (Bazel has [plans](https://docs.google.com/document/d/1d4SPgVX-OTCiEK_l24DNWiFlT14XS5ZxD7XhttFbvrI/edit#heading=h.jwrigiapdkr2) for doing this hopefully well, but it seems to be stalled at the moment.)

@fbudin69500 This was the issue that we had discussed earlier, which we weren't sure if it was due to `FindResource` logic or not. This tends to indicate that it is not.

\cc @jwnimmer-tri @sammy-tri

jamiesnape, 2018-05-31:
> With @sammy-tri's help, we found that when the example binary has srcs = ["//tools/install/libdrake:libdrake.so", ...], Bazel or the linker somehow magically realize that there are some duplications in the object code, and culls duplicate code from the resultant binary kuka_simulation.

Is this the case with both Clang and GCC? The behaviors are usually slightly different with respect to linking duplicate or unnecessary dependencies.

jwnimmer-tri, 2018-05-31:
I don't know what Bazel is doing for the magic you cite.

Maybe we should revisit the choice to use `libdrake.so` dlopen path to anchor the find_resource calls for installed Drake.  Having a stub shared library was another idea that we discussed at the time (see #7439 for `libdrake_marker.so`), and I still like that proposal better anyway.  Then `drake_example_cc_binary` dies and everything works again.

EricCousineau-TRI, 2018-05-31:
> Maybe we should revisit the choice to use libdrake.so dlopen [...]

Yup, that would solve the problem, and I assume should be relatively simple to implement. Will briefly tinker with that.

> Is this the case with both Clang and GCC?

Testing that now; will edit this post once I'm done.

PR #8909 (2018-06-04): Teach `find_resource` to use `libdrake_marker.so` in lieu of `libdrake.so`; remove `drake_example_cc_library`;
Resolves #8908 

While it would be ideal to have `find_resource` be its own library and ultimately carve up shared libraries, this is a relatively quick solution to the above issue.

\cc @jwnimmer-tri @fbudin69500 @jamiesnape 

Tested on RobotLocomotion/drake-shambhala@af8d27e

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8909)
closed 2018-06-04 by EricCousineau-TRI