RobotLocomotion/drake #8912: Create Python bindings for LcmPublisherSystem and LcmSubscriberSystem

thduynguyen, 2018-05-31:


EricCousineau-TRI, 2018-05-31:
Solution should (fingers crossed) be relatively simple; just wrap `SerializerInterface` and then implement a Python version of it, which shouldn't be too hard.

jwnimmer-tri, 2018-05-31:
I would not be surprised if the better answer was to implement a pure python version of these two systems, based on pydrake's LeafSystem.  I believe the ergonomics of that for pydrake users would be vastly improved, and the work required is approximately the same as binding all of the C++ code that these two systems require.

EricCousineau-TRI, 2018-06-01:
I think that the binding the publisher for simple cases (`AbstractValue`'s) will definitely be an equivalent effort (~5-10 lines of meaningful bindings code vs. ~5-10 of ); however, the subscriber will involve the odd synchronization mechanism.
Additionally, my view is that bindings C++ code means that I can rely on the existing test code, and just flex the API. If I write Python code anew, I am then effectively required to test that code.

All that being said, both approaches will suffer from the fact that ports of type `Value<LcmMessage>` will have trouble interacting with Python-declared ports of type `Value<py::object>`, since the Python LCM messages will not be the same.
If we want parity on ports of this type, then I will unfortunately have to register some C++ LCM types; however, that's not that big of a deal to do a small bit of finagling (maybe a 10-15 line generic function, then a single function call for each C++ LCM type that we may want port parity for).

Will give it a whirl for mebbe 30min on the binding side; if it does get hairy, then I'll def take the reimplementation route in Python.

jwnimmer-tri, 2018-06-01:
I was more imagining a python system where the user choose their own input port type (which would not be `Value<Message>`, but rather whatever plant output or thing they want to publish) as well as a python functor to convert the input to an lcm message, which the base code then serialized and transmitted.  Same thing for the subscriber.  I don't see any reason for `Value<Message>` to be involved, since I assumed the use case was "I want to publish things" and not "some system of interest already has an LCM-valued output port", since those are few and far between.

thduynguyen, 2018-06-01:
Btw, these are needed to build a simulator for our perception dev workflow. The pipeline needs an lcm subscriber to receive `lcmt_iiwa_command` messages, and an lcm publisher to publish `robotlocomotion::image_array_t`. 

On the publisher side, we also need a Python binding for `ImageToLcmImageArrayT` system to convert output images from `RgdbCamera` to lcm messages.

mpetersen94, 2018-07-22:
Has this issue been resolved? Looking to use LcmSubscriber and LcmPublisher in python as well.

EricCousineau-TRI, 2018-07-24:
Hup, sorry for the stall! Will update it.

PR #8914 (2018-07-26): pydrake systems lcm: Enable publishers and subscribers 
Resolves #8912

Took a tad bit longer than 30min, but it works at present, and keeps things closer to the C++ API while still (IMO) providing of just providing an LCM type for the serializer.

\cc @thduynguyen 

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8914)
closed 2018-07-26 by jwnimmer-tri