RobotLocomotion/drake #8918: Binary install_prereqs missing python-tk

gizatt, 2018-06-01:
Encountered by a user on Wednesday at the Cornell tutorial, and also encountered by myself when setting up a test image. Needs to be pulled from apt, I think?

jwnimmer-tri, 2018-06-05:
@gizatt What program or use case evidences the missing dependency -- `drake-visualizer` maybe?

gizatt, 2018-06-05:
I think for both of us, it wound up being a matplotlib dependency -- which shows up in the python bindings, maybe a little in VTK, and yeah, in director.

jamiesnape, 2018-06-20:
Can you be more specific about the exact piece of code that fails?

gizatt, 2018-06-20:
On another inspection, I can't find a place in the Drake binaries that specifically fails because of this without appropriate documentation. I have [downstream uses that fail without matplotlib](https://travis-ci.org/gizatt/drake_periscope_tutorial/jobs/394708619) because they depend on matplotlib for visualization, but I get the sense Drake doesn't distribute anything (other than [isolated](https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/systems/rgbd_camera_example.py) [examples](https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/systems/drawing.py) of the python bindings. The latter of those is documented to that effect (i.e. it includes a line about needing matplotlib to work), but the former isn't.

So maybe this is just working as intended?

jamiesnape, 2018-06-20:
We do include `matplotlib` in the installation scripts. Do we need `python-tk` for those examples to work? I do see that we are missing `pydot` from the installation scripts, however.

jamiesnape, 2018-06-20:
OK, so the issue is that that `python-matplotlib` and `python-tk` need to move from the source distribution installation script to the binary one. I will PR a fix momentarily.

PR #9034 (2018-06-20): Move matplotlib and tk to binary distribution prerequisities scripts and add pydot
Fixes #8918.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9034)
closed 2018-06-20 by jamiesnape