RobotLocomotion/drake #8940: Upgrade to VTK 8 / 8.1 breaks some features in director / drake visualizer

EricCousineau-TRI, 2018-06-06:
## Description

Per discussions in Drake Developer Meeting and on Slack, @manuelli and @peteflorence have been impacted by the upgrade to VTK 8 / 8.1. Per Lucas's comment:

> Yes, click + zoom with  no longer works, and puts the camera into a strange state. This is the most annoying one as I use it all the time. Also a few others as I recall, I think that attempting to add a frame a move a visualized pointcloud would also cause director to crash.

## Reproduction

@manuelli Can I ask if you can elaborate on how to reproduce these crashes?

## Debug Information

Given that this is a known repository + workflow, `spartan` used by RobotLocomotion group, skipping debug information.

manuelli, 2018-06-06:
**Issue 1**: click + f which is used to zoom and change the focus of the camera rotation no longer works.

**How to reproduce**: This can be reproduced in `spartan/master`. Note that this has an older version of directory fron Jan 26, so maybe this has been fixed in director/master when built with vtk8. We haven't had time to get around to updating, but the same procedure should work to test if this is an issue in drake-visualizer that ships with drake.

1. open director/drake-visualizer, make sure there is something in the scene that you can click on, i.e. a robot, a primitive geometry etc.
2. `left click + f` on the point you want to zoom in on. (this works just fine in the directory binaries by the way, if you want to see it in action). Currently this puts the camera into a strange state, with a weird center of focus. I end up having to hit `R` to reset the camera as the only way to recover from this.

fbudin69500, 2018-06-08:
@manuelli I was trying to reproduce your bug. I recompiled `spartan` on my computer but I couldn't reproduce your problem. I tried to use `left click +f` but it seems that both those events are separate: `left click`=" rotate the camera around its focal point" and `f` = "fly to the picked point". Is something different suppose to happen when both events happen at the same time?

manuelli, 2018-06-08:
Sorry, the I didn't explain well. It's not simultaneous. 

Step 1: Click on the point where you want to center camera rotation.
Step 2: hit `f` on keyboard

This gets me into the weird state. Use `r` to reset the camera.

manuelli, 2018-06-22:
I just finished updating `spartan` to `drake/master` and `director/master` and can confirm that the issue is still present.

EricCousineau-TRI, 2018-12-03:
Will be fixed in VTK 9, but won't be fixed prior to that.

manuelli, 2019-01-02:
Any updates as to whether this has been fixed? It's still biting us

jamiesnape, 2019-01-02:
Will be fixed in the next release of VTK, but no ETA on when that will be.

EricCousineau-TRI, 2019-01-14:
@manuelli I've been hacking on my side and tinkering with affordance markers.

With Jamie's help, (on `visualization.py`), I've applied this patch to `lib/python2.7/site-packages` in `drake-visualizer`:
```patch
diff --git a/affordanceitems.py b/affordanceitems.py
index 051bdb8..f0882ee 100644
--- a/affordanceitems.py
+++ b/affordanceitems.py
@@ -246,7 +246,7 @@ class MeshAffordanceItem(AffordanceItem):
                     transform.Scale(scale)
 
                     transformFilter = vtk.vtkTransformPolyDataFilter()
-                    transformFilter.SetInput(polyData)
+                    transformFilter.SetInputData(polyData)
                     transformFilter.SetTransform(transform)
                     transformFilter.Update()
 
diff --git a/visualization.py b/visualization.py
index bcc6aa5..59278a6 100644
--- a/visualization.py
+++ b/visualization.py
@@ -1097,7 +1097,7 @@ def pickImage(displayPoint, view, obj=None):
 
 def pickProp(displayPoint, view):
 
-    for tolerance in (0.0, 0.005, 0.01):
+    for tolerance in (0.005, 0.01):
         pickType = 'render' if tolerance == 0.0 else 'cells'
         pickData = pickPoint(displayPoint, view, pickType=pickType, tolerance=tolerance)
         pickedPoint = pickData.pickedPoint
```

Will ponder enabling this patch in `drake`; otherwise, perhaps we can just default this in our distributed version?

manuelli, 2019-01-16:
That fixed it for me, thanks Eric! I can open a PR against director master.

jamiesnape, 2019-01-16:
Note that it is a workaround rather than a fix. You are losing precision and performance in many circumstances.

manuelli, 2019-01-16:
Ok thanks for the clarification @jamiesnape

manuelli, 2019-01-16:
This fixes the `ctrl + f` but problem but now if I add a frame to a visualization object then i can no longer hide it by double clicking that frame. Just FYI, not sure if there is an easy fix for that.

jamiesnape, 2019-01-17:
> For users who have the ability to build against a patched vtk8, would you be able to link to the upstream commit so I can take a look? thanks!

I would have to confirm, but my understanding is that there was not one commit, and since it was not viable to get the patches backported to VTK 8.1, patching your own VTK is probably not going to be possible. The options are the workaround, using VTK `master`, or waiting on the next version of VTK (no ETA as far as I am aware).

EricCousineau-TRI, 2019-01-24:
Awesome, thank you! Spoke to Jamie, and we'll be able to upload the updates to Drake once your PR is merged in master. Thanks!

patmarion, 2019-01-24:
one last detail, i'm not convinced that hdpi is working exactly right.  i haven't had a chance to look into it yet, and it could just be a problem specific to my laptop, but it seems like some picking operations have a threshold that is too big, and some mouse interaction events like zoom sometimes feel like the motion factor is scaled too high.

everything still works, but it feels like i need to "retune" a few values, or investigate whether it could be qt/vtk/hdpi related.  im aware of some hdpi api in vtk but i am not familiar with it yet.

EricCousineau-TRI, 2019-03-07:
Howdy Pat! Just to follow-up, can I ask if you have any leads on resolving this?
Is there anything that @jamiesnape or I can help with?

I'd really enjoy being able to use the fixes that you've got! (Been using the affordance markers for some workflows for camera placement + IK)

patmarion, 2019-03-11:
hey, thanks for checking in!  I just merged the fixes into `master`

https://github.com/RobotLocomotion/director/pull/618

sorry i haven't been updating master more frequently, i've been putting my new work onto the python3 branch instead.  but most of it is easy to merge into `master`.  let me know if this has resolved everything for you.  I've been using this with vtk 8.2 and qt 5.12 and python3.

EricCousineau-TRI, 2019-03-11:
Sweet, thank you so much!

@jamiesnape - can I ask if you have time to submit a PR that I (and possibly @manuelli) could step through to see if these problems are resolved?

manuelli, 2019-03-11:
I can test this out on `spartan` as well and see if it fixes the problem there. Pretty swamped this week but will get to it when I have some time.

jamiesnape, 2019-03-11:
> @jamiesnape - can I ask if you have time to submit a PR that I (and possibly @manuelli) could step > through to see if these problems are resolved?

It will be up when I have resolved the OSPRay issues; so probably in the next day or two.

jamiesnape, 2019-03-12:
Try this branch: https://github.com/jamiesnape/drake/tree/vtk-8-2-0. It is not ready for PR yet, but it should be far enough along for you to try drake-visualizer.

EricCousineau-TRI, 2019-03-14:
I will try this out on my machine within the next few days for testing the affordance markers.

EricCousineau-TRI, 2019-03-15:
Just tried it out on Anzu; seems that there may be a linking error, as when it tried to import `pydrake` (necessary for me checking my workflow), it reported `libdrake.so not found`.

Going into my `.runfiles`, these are the libs that seem to be awry:
```
$ ldd external/drake/tools/install/libdrake/libdrake.so | grep 'not found' 
        libvtklzma-8.2.so.1 => not found
        libvtkdoubleconversion-8.2.so.1 => not found
```

EricCousineau-TRI, 2019-03-15:
After hacking in those two, I still have odd mouse behavior :( Not yet sure if it's due to affordance markers, or another item.

Also, point-picking is still a bit off. If I select `Pick type: raycast nearest surface point`, I can get several points that are clearly not on the surface I was clicking. If I switch to `raycast nearest vertex`, that works fine. However, now, `hardware render` works like a charm!

Are there other settings that I should tweak for the affordance markers?
(I'll see if I can post a desktop recording tomorrow.)

patmarion, 2019-03-15:
hi eric, that sounds like it might be the same picking threshold issue I have been seeing.  I think it is an issue in vtk and not director, since the director code for picking is just using basic vtk operations, but i haven't had a chance to take a closer look :/

jamiesnape, 2019-03-15:
> Just tried it out on Anzu; seems that there may be a linking error, as when it tried to import `pydrake` (necessary for me checking my workflow), it reported `libdrake.so not found`.
>
> Going into my [...]

Are the files physically missing. Does running from an install tree work?

EricCousineau-TRI, 2019-03-15:
> Are the files physically missing. Does running from an install tree work?

The files are there; here's the hack patch: https://github.com/EricCousineau-TRI/drake/commit/0f4043267edf98ebc8e520d6017fff283fd32e25

>  I think it is an issue in vtk and not director [...]

Makes sense! Is there a flag I can flip to have it use the `hardware render` style picking?

patmarion, 2019-03-15:
It depends what pick operation you are doing?  Most of the picking in the Python layer uses the function `pickProp` which does a hardware pick first.  If nothing is picked then it tries a ray cast pick with a tight tolerance and then another one with a bigger tolerance.  This operation is used, for example, when you double click or right-click on an object.

If you are talking about clicking on the frame widget, then this always does a raycast pick.  I found this operation needed to be re-tuned.  I have adjusted it in my branch but not landed that code into master because i haven't had time to fully root-cause the issue.

It seems like the ray cast picking might be having issues, so if the hardware pick doesn't make an exact match, then maybe what you are seeing is some tolerance issues with the surface picker.  Please let me know more about where you are seeing the issue.

EricCousineau-TRI, 2019-03-19:
Sorry for the late follow-up! I'll see if I can use the VTK versions in #10941, and try to port my example to a Drake branch (right now it's Anzu, TRI's version of Spartan).

EricCousineau-TRI, 2019-03-19:
Ah! Set the pick tolerance to `0` for adding measurement markers (using raycasting), and that fixed it! ([repro script](https://github.com/EricCousineau-TRI/drake/blob/8b9324162/tmp/director_marker_repro.py))
(Will see about making that default in Drake, probably through a plugin for now.)

However, I'm still struggling with the affordance markers (or rather, frame editing).
Here's a video of what the mouse behavior looks like: https://youtu.be/KojIkti5pCQ (using #10941 as the base commit)
Repro script (also in video descr.): https://github.com/EricCousineau-TRI/drake/blob/a1fdc088d/tmp/director_marker_repro.py

Is there another knob that I can tune for this to make the mouse hover functionality less... loose?

patmarion, 2019-03-19:
Thanks for sharing the video.  Yeah that's exactly the picking problem I had.  I resolved it by tuning the c++ code in vtkFrameWidgetRepresentation.cxx but I didn't put that change on `master` because it was hacky.  I think I understand the problem better now, so I'll try to update my change and put it on master.

I think the problem might be something similar to this older bug report-

https://public.kitware.com/pipermail/vtkusers/2012-June/074985.html

According to the docs, the tolerance is supposed to be in display coords, but vtkCellPicker and some other routines are using tolerance in world space.  I think maybe it's always been a problem and somehow it got worse in the latest vtk release.

patmarion, 2019-03-19:
here's the PR.  waiting for CI...

https://github.com/RobotLocomotion/director/pull/620

patmarion, 2019-03-19:
merged into `director/master` (`70c4925`)

EricCousineau-TRI, 2019-03-19:
Awesome, thank you!

jamiesnape, 2019-03-20:
Seeing compile errors on Mac:
```
Undefined symbols for architecture x86_64:
  "_PyModule_Create2", referenced from:
      _real_initvtkDRCFiltersPython in libvtkDRCFiltersPythonD.a(vtkDRCFiltersPythonInitImpl.cxx.o)
  "_PyUnicode_FromString", referenced from:
      PyvtkFrameWidgetRepresentation_GetBounds(_object*, _object*) in libvtkDRCFiltersPythonD.a(vtkFrameWidgetRepresentationPython.cxx.o)
      PyvtkLCMGLProp_GetBounds(_object*, _object*) in libvtkDRCFiltersPythonD.a(vtkLCMGLPropPython.cxx.o)
ld: symbol(s) not found for architecture x86_64
```

jamiesnape, 2019-03-20:
```
cmake \
  -DBUILD_SHARED_LIBS:BOOL=OFF \
  -DCMAKE_BUILD_TYPE:STRING=Release \
  -DCMAKE_FIND_FRAMEWORK:STRING=LAST \
  -DCMAKE_INSTALL_PREFIX:PATH=/opt/drake \
  -DCMAKE_PREFIX_PATH:PATH=/opt/drake \
  -DDD_QT_VERSION:STRING=5 \
  -DPYTHON_EXECUTABLE:PATH=/usr/local/bin/python2 \
  -DUSE_EXTERNAL_INSTALL:BOOL=ON \
  -DUSE_LCM:BOOL=ON \
  -DUSE_LCMGL:BOOL=ON \
  -DUSE_SYSTEM_EIGEN:BOOL=ON \
  -DUSE_SYSTEM_LCM:BOOL=ON \
  -DUSE_SYSTEM_LIBBOT:BOOL=ON \
  -DUSE_SYSTEM_VTK:BOOL=ON \
  -DVTK_DIR:PATH=/usr/local/opt/vtk@8.2/lib/cmake/vtk-8.2 \
  ../director/distro/superbuild
```
where `/opt/drake` is an unpacked 
https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-mac.tar.gz and `/usr/local/opt/vtk@8.2` is from `brew install robotlocomotion/director/vtk@8.2`.

EricCousineau-TRI, 2019-03-20:
Hm... `_PyModule_Create2` and `_PyUnicode_FromString` smell like Python version mismatch linker errors...

jamiesnape, 2019-03-20:
They do, but not sure why they are occurring with this version and not the previous one.

EricCousineau-TRI, 2019-03-20:
If you built VTK in Mac with Python 2 + Python 3 (in that order), perhaps it's due to some criss-crossing of install artifacts (maybe CMake config files?) that makes director partially pick up Python 3 modules when trying to compile as Python 2?

Not sure why it's new though...

jamiesnape, 2019-03-20:
Ok, I think it was user error. We should be good now.

jamiesnape, 2019-03-21:
Closed by #10941.

closed 2019-03-21 by jamiesnape

EricCousineau-TRI, 2019-03-21:
@manuelli If you still encounter the issues you had before (key bindings, etc.), please post here and we can re-open the issue.

Thanks!