RobotLocomotion/drake #8942: Perfomance profiling of the Implicit Stribeck in MBP

amcastro-tri, 2018-06-06:
## Description
I performed profiling of the [simple_gripper](https://github.com/amcastro-tri/drake/tree/implicit_stribeck_in_mbp/examples/simple_gripper) example using the implicit Stribeck time-stepping method in MBP (Implicit Stribeck PR #8925 under review, you can reproduce this profiling off my dev branch with everything integrated into MBP [here](https://github.com/amcastro-tri/drake/tree/implicit_stribeck_in_mbp/examples/simple_gripper)).

In a nutshell the summary of costs looks something like this:
- `MultibodyPlant::DoCalcDiscreteVariableUpdates()`: 96.34%
- `ProximityEngine::ComputePointPairPenetration()`: %60.22
- `MultibodyPlant::CalcTangentVelocitiesJacobian()`: 17.98%
- `MultibodyTree::CalcMassMatrixViaInverseDynamics()`: 9.04%
- `ImplicitStribeckSolver::SolveWithGuess()`: 2.82%
- `Eigen::LDLT::compute()`: 0.32%

Some observations:
- between 5%-10% improvement in performance (I might be underestimating here) could be attained with caching in MBP for the Jacobian computations.
- Clearly there is no need to improve performance of the contact solver at this point.
- Cholesky factorizations comprise a negligible amount of the cost (you were right @sherm1!, of course).
- FCL is eating up a lot of the cost, though **there is no filtering** in this case and **visuals are going trhough FCL's pipeline** as well.

## Reproduction

- Checkout the [`simple_gripper` example from my dev branch](https://github.com/amcastro-tri/drake/tree/implicit_stribeck_in_mbp/examples/simple_gripper).
-  Follow the instructions in the `README` file to build the example and on how to navigate to the `.runfiles` directory.
- cd into the runfiles directory `cd bazel-bin/examples/simple_gripper/simple_gripper.runfiles/drake`.
- Run under `callgrind`: 
    -  `valgrind --tool=callgrind ./examples/simple_gripper/simple_gripper --simulation_time=20.0 --v_stiction_tolerance=1e-4 --target_realtime_rate=0 --amplitude=0.05 --frequency=2.0`
- Last step will generate a file named `callgrind.out.some_number`
- Visualize statistics with `qcachegrind callgrind.out.number_from_previous_step`

With the default time step of `1.0e-3` secs this will run `20,000` time steps, good enough to get statistics.

amcastro-tri, 2018-06-06:
cc'ing @SeanCurtis-TRI. Hopefully this helps you reproduce it.

amcastro-tri, 2018-06-08:
A little update. When completely removing the visuals (commented out in the SDF's for both the gripper and the mug) for this example the costs get split as:
- MultibodyPlant::DoCalcDiscreteVariableUpdates(): 94.5%
- ProximityEngine::ComputePointPairPenetration(): 38.8%
- MultibodyPlant::CalcTangentVelocitiesJacobian(): 29.2%
- MultibodyTree::CalcMassMatrixViaInverseDynamics(): 14.8%
- ImplicitStribeckSolver::SolveWithGuess(): 4.0%
- Eigen::LDLT::compute(): 0.5%

This still **does not apply filtering for geometries on the same body**. Particularly important for this sim case since each finger has 8 collision geometries (ring of spheres for the pads).
Therefore, there will be a considerable speed up as filtering lands in SG.

SeanCurtis-TRI, 2018-06-08:
FTR Would it be correct to say that in the list of functions you've shown above, that `MultibodyPlant::DoCalcDiscreteVariableUpdates()` invokes all of the other functions listed? Also, further elaboration on the percentages would be good -- as they don't total to 100%, one must infer that they reflect the percentage of the level of the call tree in which they are invoked. So, understanding that something is x% of some other thing that is y% of the overall process will actually give us a better sense of what cost we're paying.

jwnimmer-tri, 2018-06-08:
The percentages mean "what fraction of total execution time was spent with this function on the call stack somewhere".  So `main()` (if shown) would be nearly 100%, for example.  I think the important line item above is that the `ComputePointPairPenetration` workload is ~40% of total runtime, so if we can perhaps decimate its workload, we could gain a ~50% improvement in throughput.

jwnimmer-tri, 2018-06-08:
Ah, sure; I assumed they were global percentages -- though yes absolutely if they are not global then that should be clarified.

amcastro-tri, 2018-06-08:
@jwnimmer-tri is correct. I updated the description.

amcastro-tri, 2018-06-12:
Update on the profiling incorporating geometry filtering (PR #8954). Even slightly better than removing all visuals in the last profile. (runs at 3.5X in my HP Z640).

- MultibodyPlant::DoCalcDiscreteVariableUpdates(): 93.6%
- ProximityEngine::ComputePointPairPenetration(): 34.17%
- MultibodyPlant::CalcTangentVelocitiesJacobian(): 31.7%
- MultibodyTree::CalcMassMatrixViaInverseDynamics(): 15.86%
- ImplicitStribeckSolver::SolveWithGuess(): 4.25%
- Eigen::LDLT::compute(): 0.6%

sherm1, 2018-06-12:
Can't tell from these percentages how much the overall runtime is improving -- can you give a total time or rate vs. real time?

amcastro-tri, 2018-06-12:
The first test listed at the top, run at about 2X. With filtering we go up to 3.5X (in my (slow) local workstation). Time step is 1ms.

amcastro-tri, 2018-09-11:
[off master](https://github.com/RobotLocomotion/drake/tree/master) | [commit](https://github.com/RobotLocomotion/drake/commit/22d2577ca03fb7e667942e09fe02e726f32720e4)

These are **global percentages**.

- MultibodyPlant::DoCalcDiscreteVariableUpdates(): 91%
- MultibodyPlant::CalcTangentVelocitiesJacobian(): 24%
- MultibodyPlant::CalcNormalSeparationVelocitiesJacobian(): 23%
- ProximityEngine::ComputePointPairPenetration(): 20%
- MultibodyTree::CalcPositionKinematicsCache(): 14%
- MultibodyTree::CalcMassMatrixViaInverseDynamics(): 12%
- ImplicitStribeckSolver::SolveWithGuess(): 6.3%
- Eigen::LDLT::compute(): 0.4%

Immediate action item described in #9428 to speed up Jacobians computation.

amcastro-tri, 2018-09-12:
[off master](https://github.com/RobotLocomotion/drake/tree/master) | [commit](https://github.com/RobotLocomotion/drake/commit/ccb13358d72d638b8a3da68bf60c71afb380812b)

(Right after the merging of #9433 to factor out common computations for the contact Jacobians.)

These are **global percentages**.

- MultibodyPlant::DoCalcDiscreteVariableUpdates(): 88%
- MultibodyPlant::CalcNormalAndTangentContactJacobians(): 32%
- ProximityEngine::ComputePointPairPenetration(): 26%
- MultibodyTree::CalcMassMatrixViaInverseDynamics(): 15%
- MultibodyTree::CalcPositionKinematicsCache(): 9.7%
- ImplicitStribeckSolver::SolveWithGuess(): 8.2%
- Eigen::LDLT::compute(): 0.5%

SeanCurtis-TRI, 2018-09-12:
A passing thought. It'd be cool if you could add an additional metric -- which is overall cost. In other words, for a given simulation (with given integrator settings), `m` seconds of simulation takes `n` seconds of wall time to run. And *then* the breakdown. With these percentages, it's not clear if anything gets faster or if cost is just reapportioned.

amcastro-tri, 2018-11-29:
[off master](https://github.com/RobotLocomotion/drake/tree/master) | [commit](https://github.com/RobotLocomotion/drake/commit/65a5322a2ce1ccb9585a0383d9439dbd495aac00)

These are **global percentages**.

MultibodyPlant::DoCalcDiscreteVariableUpdates(): 86%
ProximityEngine::ComputePointPairPenetration(): 31%
MultibodyTree::CalcMassMatrixViaInverseDynamics(): 19%
MultibodyPlant::CalcNormalAndTangentContactJacobians(): 17%
ImplicitStribeckSolver::SolveWithGuess(): 10.5%
Eigen::PartialPivLU::compute(): 0.6%
MultibodyTree::CalcAcrossNodeGeometricJacobianExpressedInWorld(): 0.48%
MultibodyTree::CalcPositionKinematicsCache(): 0.33%

Compared to the previous profiling from Sep 12 2018, these numbers show the improvement in performance caused by caching position kinematics and across node Jacobians.
Now the cost of computing contact Jacobians and the mass matrix are comparable.

As a reference, in my HP Z640 this runs at (parameters described at the very top):

4.94 X: with **no** visualizer running.
4.10 X: **with** visualizer running.

amcastro-tri, 2019-03-11:
[off master](https://github.com/RobotLocomotion/drake) | [commit](https://github.com/RobotLocomotion/drake/commit/47987499486349ba47ece6f30519aaf8f868bbe9)

As a reference, in my HP Z640 this runs at (parameters described at the very top):

8.5 X: with **no** visualizer running.
7.0 X: **with** visualizer running.

There has been a significant speed bump for this case. I will update soon with the breakdown to understand what caused it.

sherm1, 2019-03-11:
It took me some time to understand those numbers -- I thought you were reporting a slowdown! Most people have been reporting speed as `simtime / realtime` so _smaller_ is better. I _think_ you are reporting `realtime / simtime` so _larger_ is better. To avoid confusion, I'd suggest either reporting the reciprocal or always accompanying the reported number with something like `(realtime / simtime, larger is better)`.

sherm1, 2020-05-11:
@amcastro-tri is it still worth having this open?

amcastro-tri, 2020-05-13:
I'll keep open to track the speedups.

SeanCurtis-TRI, 2021-12-04:
Looks like 1.5 years since "speedups" have been tracked. I'm reassigning this to you @amcastro-tri and you can close it if you like.

amcastro-tri, 2021-12-06:
I think this reached the end of its lifetime. Thanks @SeanCurtis-TRI

closed 2021-12-06 by amcastro-tri