RobotLocomotion/drake #8943: Reconcile "anchored" geometry between MBP and SceneGraph

SeanCurtis-TRI, 2018-06-06:
## Description

SceneGraph distinguishes between two types of geometry: anchored and dynamic. The distinction is one of efficiency; for geometry that is "anchored" to the world:
  - poses do not need to be reported at each update
  - we can automatically exclude collisions between anchored geometry
  - we can support more complex geometries (i.e., anchored geometries can use expensive pre-computation techniques to accelerate queries because it would only need to be done once. A corresponding dynamic geometry would have to recompute at each update.

The syntactic distinction between anchored and dynamic geometry is that anchored geometry are added through an explicit interface which associates the geometry implicitly with SceneGraph's world frame. Registering dynamic geometry requires registering a GeometryFrame first and then registering the geometry on that frame. The geometry source (in this case, the MBP) *owns* each registered GeometryFrame and for each registered GeometryFrame, the geometry source must provide a pose at every update call. Thus, anchored geometry requires no `FrameId` or the concomitant pose updates; dynamic geometry does.

MultibodyPlant can introduce anchored geometry in a couple of different ways:
  - Explicit: invoke `MultibodyPlant::RegisterGeometry()` (which registers geometry onto a body), with the world body the body parameter. This will directly invoke the API for registering anchored geometry (current feature)
  - Implicit: geometry registered to a body that has a rigid path to the world *should* also be considered anchored (future feature).

The typical workflow in MultibodyTree is to build the tree component by component. I.e., add a body, then add the geometries attached to that body. This produces a `FrameId` for the body and a `GeometryId` per registered geometry. However, only after `MultibodyTree::Finalize()` is called will the tree know if a body has a rigid path to the world. At that point, the FrameId is now misleading and the geometry is functionally anchored.

### The Issue
In order for the MBP to work in this "construct-then-finalize" approach, it needs a mechanism to be able to change a dynamic geometry to anchored. As a related feature, it should be able to dispose of a `FrameId`, with the attached geometry being converted to anchored geometry. However, it must maintain some sense of what body it was "attached" to. Querying the geometry what body it belongs to should ultimately map to the body to which it was truly assigned. In other words, there *should* be a distinction between a geometry that was explicitly declared anchored vs one that ended up being implicitly anchored.

sherm1, 2020-06-15:
@SeanCurtis-TRI is this still an issue?

SeanCurtis-TRI, 2020-06-15:
The issue still exists. Perhaps I should ask, do we care? Right now, any body that gets welded is still considered "dynamic" from SG's perspective. MBP must provide poses for them and SG is still keeping them in the BVH for dynamic objects and will try to bang them against geoemetries that have been *officially* affixed to the world body. So, do we care?

sherm1, 2020-06-16:
Probably the first thing to do would be to measure whether there is something to be gained by proper classification of anchored geometry. Seems plausible -- let's leave it at backlog priority.

jwnimmer-tri, 2024-03-16:
As we scale up the amount of anchored geometry in our typical scenes, it might be time to do some benchmarking to get a sense for how much this matters.

rpoyner-tri, 2026-11-11:
Reassigning to @SeanCurtis-TRI, who is driving this through #25150 (the first of three PRs, per its description). I'm not working on it. Please unassign yourself or push back if that's wrong.