RobotLocomotion/drake #8948: QueryObject<AutoDiffXd> and MBP<AutoDiffXd> support

SeanCurtis-TRI, 2018-06-07:
## Description

`MultibodyPlant<T>::CalcAndAddContactFOrcesByPenaltyMethod` throws an exception by default. It has a specialization for `T = double`. The reason for this is that the result of the `QueryObject<T>::ComputePointPairPenetration()` always returns a vector of `PenetrationAsPointPair<double>` (due to incomplete autodiff support through FCL.)

## Analysis

It may be that the specialization is unnecessary. Because doubles are interoperable with the other scalar types, it would be acceptable to compute `AutoDiffXd` values from the double values stored in the collision result. The implication is that if any of those values *depend* on the gradient variables, we will lose gradients. There *is* a distinction between the compiler issue and the mathematical issues.

## Proposal

It may be sufficient to define the MBP method for all scalar types by exploiting the double-valued penetration results. This should be valid from the compiler's perspective. Mathematically, this may be resolved by adding a flag similar to that introduced in corresponding functionality [`RigidBodyTree`](https://github.com/RobotLocomotion/drake/blob/master/multibody/rigid_body_tree.h#L1054) until we have scalars all the way through the collision queries.

/cc @amcastro-tri

amcastro-tri, 2018-06-07:
I actually have I PR were I attempt to clean this up and I believe I do what you suggest. I'll see to push it next for you to review @SeanCurtis-TRI, I wonder if I got it right.

SeanCurtis-TRI, 2019-05-16:
@amcastro-tri I think we should close this issue. This topic was likewise discussed in #11120 with a very [strong conclusion](https://github.com/RobotLocomotion/drake/issues/11120#issuecomment-479093116) that seems to undermine what this proposes (i.e., let the derivatives get lost). The conclusion is that if we can't produce real answers, we shouldn't produce answers at all.

If you concur, please close this.

amcastro-tri, 2019-05-16:
Not sure if we should close. Did we already address the solution proposed [here](https://github.com/RobotLocomotion/drake/issues/11120#issuecomment-479091723)?

SeanCurtis-TRI, 2019-05-16:
The link I linked to *is* the solution you linked to. It was me distilling it. Both, however, contradict what is in this issue.

amcastro-tri, 2019-05-17:
In that case I think the solution would be to close this issue and open a new one to track the solution stated in those comments? what do you think?

SeanCurtis-TRI, 2019-05-17:
Agreed

SeanCurtis-TRI, 2019-05-17:
Closing in favor of #11120.

closed 2019-05-17 by SeanCurtis-TRI