RobotLocomotion/drake #8960: pydrake eigen: Remove check semantics to more accurately reflect C++

EricCousineau-TRI, 2018-06-08:
Per comment:
https://github.com/robotlocomotion/drake/pull/8956#issuecomment-395849867

To answer your question: I had started it out this way in wanting to have the Python functionality be a little more robust, but it does create awkward asymmetries, in that you can pass a non-unit / orthonormal `Eigen/Geometry` element from C++ to Python, and it won't know any better.

Ideally, the geometric entities in `drake/math/...` could be used to be better leverage checking for rotation matrices. For quaternions and angle-axes, I'm not sure what would be used in its place, but perhaps an overloaded function in Python such as `IsUnit()`.

\cc @SeanCurtis-TRI

EricCousineau-TRI, 2019-05-08:
Moving code to issue:
```
Do not require only SE(3) for isometries because this
makes it a misnomer: SE(3) ⊂ Isometry(3) b/c Isometry admits reflection.
Additionally, do not validate for Quaternion, b/c it will break parity with
C++ and cause odd edge cases when attempting to do projections from non-unit
quaternions to unit quaternions (e.g. for use with `RigidTransform`).
```

EricCousineau-TRI, 2020-05-02:
Lowering priority for now.

mpetersen94, 2021-03-18:
I've run into this problem repeatedly in my code as I try to do calculations based on the quaternion I get out of a MBP Simulator.  I keep having to write something similar to 
```
x = mbp.GetPositions(context)
quat = Quaternion(x[:4] / np.linalg.norm(x[:4]))
```

Is the solution to just delete [this function](https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/common/eigen_geometry_py.cc#L62-L68) and all the places it is called in the bindings?  Or is there a better solution?

EricCousineau-TRI, 2021-03-18:
Yup, and sorry about that! Filing PR now.

sherm1, 2021-03-18:
BTW the Drake Simulator doesn't fix quaternion drift in the state (although it should). Instead it normalizes prior to use so that the results are independent of the quaternion norm. (The quaternion time derivatives are computed using the unnormalized quaternions so they are scaled correctly.) Surprisingly I don't see an issue about that -- I'll file one.

EricCousineau-TRI, 2021-03-22:
I hadn't read closely enough; in Mark's case above, the check was actually kind of a useful thing (IMO).

Given that Sherm filed #14800, I'm going to go ahead and close this issue for now; it can be reopened if there is a concrete pain point in the future.

(Mark, feel free to counter here or in the PR)

closed 2021-03-22 by EricCousineau-TRI