RobotLocomotion/drake #8964: drake_visualizer import error: no module named vtkRenderingOSPRayPython

RobotXiaoFeng, 2018-06-10:
## Description

Hi, I am new to drake and am trying to run the kuka_iiwa example here: https://github.com/RobotLocomotion/drake/tree/master/examples/kuka_iiwa_arm. I build drake using bazel then follow the instructions in the example. However, when I try to launch drake_visualizer, there is an 

> File "/home/drake/bazel-bin/tools/drake_visualizer.runfiles/drake/external/vtk/lib/python2.7/site-packages/vtk/vtkRenderingOSPRay.py", line 9, in <module>
    from vtkRenderingOSPRayPython import *
ImportError: No module named vtkRenderingOSPRayPython

It seems that a vtkRenderingOSPRayPython.py is missing? I found a vtkRenderingOSPRayPython.so in that path.

And it seem that the pre-compiled package has the same problem.

jwnimmer-tri, 2018-06-10:
Possibly related to #8789.

However, 
```console
$ bazel build //tools:drake_visualizer
$ bazel-bin/tools/drake_visualizer
```
works just fine for me on Ubuntu 16.04, using latest master.

Please fill out the full bug report template that was provided when you opened the issue.  If we can't reproduce the problem (what commands did you run? what OS? what revision of Drake? what is your PYTHONPATH?), we can't help fix it.

RobotXiaoFeng, 2018-06-10:
I've fill the bug report template.

EricCousineau-TRI, 2018-06-11:
Thanks for using the template!
Just out of curiosity, can I ask if you can post the full stack trace to that import error? I'm curious which module in VTK Python is requesting OSPRay at this point.

Also, I see that you may be mixing up the Python bindings build; at present, you won't be able to import `pydrake` from the Bazel-ran version of `drake_visualizer`; since you are using the binary package from 2018-02-05 but a commit from the past few days, there may be some weird stuff going on with version mixing.

Can you try doing `bazel run //tools:drake_visualizer` in a fresh terminal, with no extra `PYTHONPATH` set?

If that still does not work, can you try running the command with a completely fresh environment?
```
$ alias bash-isolate='env -i HOME=$HOME DISPLAY=$DISPLAY SHELL=$SHELL TERM=$TERM USER=$USER PATH=/usr/local/bin:/usr/bin:/bin bash --norc'
$ bash-isolate
bash-4.3$ cd drake
bash-4.3$ bazel run //tools:drake_visualizer
```

RobotXiaoFeng, 2018-06-12:
Hi Eric,
I've post the full stack trace~

`bazel run //tools:drake_visualizer` in a fresh terminal without any extra set also fails. 

And 
> $ alias bash-isolate='env -i HOME=$HOME DISPLAY=$DISPLAY SHELL=$SHELL TERM=$TERM USER=$USER PATH=/usr/local/bin:/usr/bin:/bin bash --norc'
$ bash-isolate
bash-4.3$ cd drake
bash-4.3$ bazel run //tools:drake_visualizer
gives the same error as well.

I moved from windows to ubuntu only recently, maybe this problem is due to some inappropriate settings? 

Also, the binary is downloaded and try after I get this error with the build using bazel, so I think maybe it is not due to a mix use of different versions?

mws262, 2018-06-12:
Another coworker and I are having this issue too. I found a stupid hack to get around it. I imagine altering things in the bazel-bin folder is evil, but maybe this will help you diagnose the true root of the issue.

The error happens .../drake/bazel-bin/tools/drake_visualizer.runfiles/drake/external/vtk/lib/python2.7/site-packages/vtk/vtkRenderingOSPRay.py:

```
from __future__ import absolute_import
try:
    # use relative import for installed modules
    from .vtkRenderingOSPRayPython import *
except ImportError:
    # during build and testing, the modules will be elsewhere,
    # e.g. in lib directory or Release/Debug config directories
    from vtkRenderingOSPRayPython import *
```

Replace with something like:

```
from __future__ import absolute_import
try:
    # use relative import for installed modules
    from .vtkRenderingOSPRayPython import *
except ImportError:
    print "Oh no, vtkRenderingOSPRayPython cannot load. Oh well."
```

RobotXiaoFeng, 2018-06-12:
@mws262 LOL

mws262, 2018-06-13:
This issue and several others resolved for me when I downgraded CLion from a 2018 version to 2017.3.
The Bazel plugin for CLion has not yet been updated for 2018.

RobotXiaoFeng, 2018-06-19:
This issue resolved for me after I reinstalled lightdm, however I have not figured out why.

EricCousineau-TRI, 2018-06-21:
If y'all are able to, can y'all post the output of the following commands?
```
cd drake
bazel build //tools:drake_visualizer
( cd bazel-bin/tools/drake_visualizer.runfiles/drake/external/vtk/lib && LD_LIBRARY_PATH=${PWD} ldd python2.7/site-packages/vtk/vtkRenderingOSPRayPython.so | sort | sed 's#/home/.*\.runfiles/#{runfiles}/#g' )
```
An example from my system:
https://gist.github.com/EricCousineau-TRI/a9517021eba7f51a30d0fa8c1d3c9f11#file-drake_8964_output-txt

@jamiesnape Can I ask if you know what may be happening here?

As an aside, you can automate the @mws262's hack if you apply the following patch:
https://gist.github.com/EricCousineau-TRI/a9517021eba7f51a30d0fa8c1d3c9f11#file-drake_8964_workaround-patch

RobotXiaoFeng, 2018-06-21:
Mine here: 
https://gist.github.com/RobotXiaoFeng/cbcd499ebe5d233e69dc49d1e9c15710#file-drake_8964_output-txt

kuindersma, 2018-06-21:
https://gist.github.com/kuindersma/e04fa74fbe26ddbcf70816b6389a2747#file-drake-viz-output-txt

jamiesnape, 2018-06-21:
@kuindersma `sudo apt install libtbb2`

kuindersma, 2018-06-21:
@jamiesnape that fixed it. Thx!

PR #9042 (2018-06-22): setup/ubuntu: Install `libtbb2`
Resolves #8964

(Assuming `libtbb2` was the only missing dependency.)

This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/9042)
closed 2018-06-22 by jamiesnape