RobotLocomotion/drake #8977: New constructor for Monomial class

frankpermenter, 2018-06-13:
Take drake::symbolic::Variables and Eigen::VectorXi as input and construct a Monomial object.

Ex: 
```
Monomial( vars , exponent)

where vars = ("x", "y", "z") and exponent = (1,3,4).
```

jwnimmer-tri, 2018-06-13:
Since `Variables` is unsorted, I think we instead need `VectorX<Variable>` and `VectorXi` as the two arguments.

soonho-tri, 2018-06-13:
@jwnimmer-tri, I agree. The only caveat is that `Variables` is actually ordered by `Variable::Id`. But I agree that this ordering is highly likely not the one that users expect.

@frankpermenter , could you update the issue? Also, please add the reverse operation on the `symbolic::Polynomial` to the first comment.

frankpermenter, 2018-06-13:
The "reverse" operation that takes a Monomial (or Polynomial) to a VectorXi (or MatrixXi) of exponents would also be useful.

> Ex: convert the Monomial x^2 y to [2,1]  and the polynomial x y^2 z^3 + z^2 to 
[1,2,3
 0,0,2]

soonho-tri, 2018-06-13:
@frankpermenter , can you either post your working branch here or send the link to me?

frankpermenter, 2018-06-13:
@soonho-tri 
See https://github.com/frankpermenter/drake/blob/monomialbasis/solvers/sos_utils.cc

The function "ExponentsToMonomials" implements the requested Monomial constructor and the function "GetPolynomialExponents" implements the reverse operation.

PR #8983 (2018-06-15): Add new constructor for symbolic::Monomial
Close #8977

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8983)
commit 7bc6d325 (2018-06-15): Add new constructor for symbolic::Monomial

  Monomial(const Eigen::Ref<const VectorX<Variable>>& vars,
           const Eigen::Ref<const Eigen::VectorXi>& exponents);

Close #8977
closed 2018-06-15 by sherm1