RobotLocomotion/drake #8978: vtk ospray: Synchronization issues cause segfaults ~10% of the time

EricCousineau-TRI, 2018-06-13:
## Description

Relates #8939

With the current setup of VTK's OSPRay integration, it seems that there are synchronization issues (still unsure if it's VTK, ospray, embree, or something else) that cause @kunimatsu-tri's unittests using `RgbdRendererOSPray` to segfault maybe ~10% of the time.

\cc @jamiesnape @kunimatsu-tri 

## Reproduction

Code:
* [C++ Code, consolidated](https://github.com/EricCousineau-TRI/drake/blob/486a063/tmp/rgbd_ospray_minimal.cxx#L240-L255)
* [Shell Script for both known stacktraces](https://github.com/EricCousineau-TRI/drake/blob/486a063/repro.sh) - note that build setup currently uses a local path in `tools/workspace/vtk/repsository.bzl`
* [Scripts to build VTK with OSPRay](https://github.com/EricCousineau-TRI/vtk_ospray_build/tree/f6c91d0) on the host (useful to see source paths when debugging); see a rendition of Jamie's [Dockerfile for deployment](https://github.com/EricCousineau-TRI/drake/tree/486a063bf8540145a407bf9229e54b829264ef49/tools/workspace/vtk/packaging)

Sample stacktraces:
* `NoBodyTest`: https://gist.github.com/EricCousineau-TRI/b34775b317c404a9e12479d87193edb7#file-vtk_ospray_build_stacktrace_repro-txt-L180-L194
* `MeshTest`: https://gist.github.com/EricCousineau-TRI/b34775b317c404a9e12479d87193edb7#file-vtk_ospray_build_stacktrace_repro-txt-L521-L572

Commits:
* Current head of test branch: 486a063
* Test branch's base with `r1` or `r2` of #8939: 87522d1
* `git merge-base upstream/master HEAD`: 935b4c5

VTK + OSPRay configurations tested:
* Current build binary (`master`)
* VTK v8.1.1, OSPRay v1.5.0, embree v2.17.4 - all from source, debug and release mode
* VTK v8.1.1 (patching OSPRay code with code from vtk@master), OSPRay v1.5.0, embree v2.17.4 - all from source, debug mode
* Briefly try VTK v9.0, OSPRay `master`, embree `master` - all from source
* Dockerfile build in debug mode

(There were some other configurations, but did not record them.)

## Debug Information

* What operating system are you running Drake on?
(e.g. Ubuntu 16.04, macOS 10.13)

    Ubuntu 16.04

* Is this an issue with (a) C++, (b) Python, or (c) the distributed tools
(e.g. `drake-visualizer`)?

    C++

* Are you (a) building `drake` from source, or (b) from a binary release?

    Source

* What version of `python` are you using?

    N/A

* If you are building from source or using the `drake` C++ library as an
external:

    * What version of Bazel and CMake are you using?

    ```
    + which bazel
    /usr/bin/bazel
    + bazel version
    Build label: 0.12.0
    Build target: bazel-out/k8-opt/bin/src/main/java/com/google/devtools/build/lib/bazel/BazelServer_deploy.jar
    Build time: Tue Aug 4 01:22:27 +50246 (1523462001747)
    Build timestamp: 1523462001747
    Build timestamp as int: 1523462001747
    + which cmake
    /usr/bin/cmake
    + cmake --version
    cmake version 3.5.1

    CMake suite maintained and supported by Kitware (kitware.com/cmake).
    ```

    * If you are using Bazel, please provide the host settings in your present
    project workspace:

    ```
    $ bazel run --compiler=gcc-5 @drake//common:print_host_settings
    + capture_cc_env=external/drake/common/capture_cc.env
    + source external/drake/common/capture_cc.env
    ++ BAZEL_CC=/usr/bin/gcc-5
    + /usr/bin/gcc-5 --version
    gcc-5 (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
    Copyright (C) 2015 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
    ```

    * If you are using CMake, please provide the compilers being used in your
    project:

    ```
    + /usr/bin/g++ --version
    g++ (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
    Copyright (C) 2015 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
    + /usr/bin/gcc --version
    gcc (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
    Copyright (C) 2015 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
    ```

jwnimmer-tri, 2018-06-13:
I'd also suggest trying valgrind and/or asan, to get better diagnostics.

jamiesnape, 2018-06-13:
`Dockerfile` for building VTK: https://gist.github.com/jamiesnape/78e86517130738eb60123fae2a57e2a8

kunimatsu-tri, 2018-06-13:
This is great. Thanks @EricCousineau-TRI for your digging and consolidating!

EricCousineau-TRI, 2018-06-13:
I had briefly tried using `valgrind` under GCC and was able to get some output for it; however, I had difficulty interpreting the results and distinguishing what may be SIMD efficiency-related stuff vs. what actually pointed to synchronization issues or what not.
Definitely worth digging into more deeply, though.

For ASan / TSan type stuff, I had not yet tried compiling VTK + deps with these flags to enable these sanitizers.

EricCousineau-TRI, 2019-03-07:
Per f2f w/ Jamie, he's in the process of upgrading VTK + OSPRay + embree.

EricCousineau-TRI, 2019-03-14:
Per f2f:

Have binaries for this; things working on Mac, but still needs to test on Ubuntu (Xenial + Bionic).

Summary of potential fix:
* VTK used to unconditionally call OSPRay's Init, but never shutdown.
* This changed in 8.2 to unconditionally shutdown.
  * Shutdown unconditionally causes issues.
  * VTK upstream has commit to do ref-counting. This hopefully will be upstreamed to OSPRay.
* This workflow (multiple instances of OSPRay) is unique, and never considered / exercised in the implementation of OSPRay / testing in ParaView integration.

EricCousineau-TRI, 2019-03-28:
Remaining item is to move some code in Drake, then this issue can be resolved.

tri-ltyyu, 2019-04-11:
This should be lower priority now because of upstream fixes. Putting this back to backlog for now.

ggould-tri, 2019-05-23:
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-bionic-clang-cmake-nightly-debug/185/ suspected instance.

ggould-tri, 2019-05-23:
(sorry for the accidental wrong-button-push close)

EricCousineau-TRI, 2019-05-23:
Per f2f, @jamiesnape can debug more faithfully with a physical Bionic machine. Has some potential updates.

Current rate is more like ~1%, plans to make it more like 0.01% (or 0%?) for unittests using a mutex.

jamiesnape, 2019-07-23:
Most relevant upstream fix was https://github.com/Kitware/VTK/commit/c1911ec919bb4119cee7b572f38f6da8a9deb06e. OSPRay support has been completely refactored since and additional (similar) fixes have been applied.

SeanCurtis-TRI, 2020-09-25:
This is about `RgbdRendererOspray`, that class is no no more. I propose we close this. We haven't encountered the same issues with `RenderEngineOspray`, yes?

EricCousineau-TRI, 2020-09-25:
I think this apply would apply equally to `RenderEngineOspray` due to the singleton bits.
I would say this would be 100% closable if this `flaky` tag were removed:
https://github.com/RobotLocomotion/drake/blob/868c758afb89c0e25802a99335688d9cf617dd72/geometry/render/BUILD.bazel#L211-L212

SeanCurtis-TRI, 2020-09-28:
Yep. You're right. It stays open.

jwnimmer-tri, 2020-12-10:
Resolved in #14339 by removing RenderEngineOspray entirely.

closed 2020-12-10 by jwnimmer-tri