RobotLocomotion/drake #8991: stateless system with a fixed step integrator does not generate publish events.

RussTedrake, 2018-06-14:
Related to #8990.  

Adding the following lines to `signal_logger_test.cc` (#8990 contains the missing #include and namespace update) reveals what I consider incorrect behavior (which surfaced in https://stackoverflow.com/questions/50784466/child-class-of-leafsystem-generating-sinusoidal-signal/50795952?noredirect=1#comment88634316_50795952).  

```
// Ensure that we can log a constant signal by specifying
// publish_every_time_step.
GTEST_TEST(TestSignalLogger, PublishEveryTimeStep) {
  // Add System and Connect
  DiagramBuilder<double> builder;
  auto system = builder.AddSystem<ConstantVectorSource<double>>(2.0);
  auto logger = LogOutput(system->get_output_port(), &builder);
  auto diagram = builder.Build();

  // Construct Simulator
  systems::Simulator<double> simulator(*diagram);
  simulator.set_publish_every_time_step(true);
  simulator.get_mutable_integrator()->set_maximum_step_size(0.1);

  // Run simulation
  simulator.StepTo(1);

  std::cout << logger->sample_times() << std::endl;
  EXPECT_EQ(simulator.get_num_publishes(), 11);
}
```

The above test fails.  It only publishes at time=0 (and does that twice).  I would expect this to be sufficient to result in periodic publish events.  

cc @sammy-tri , @EricCousineau-TRI

edrumwri, 2018-10-08:
It _appears_ that you've registered another issue that seeks exactly the opposite behavior (#4376). Am I reading that wrong? The original Github link is no longer valid so I can't be sure.

RussTedrake, 2018-10-09:
I think @EricCousineau-TRI and I believe that set_publish_every_time_step should live at the System level, not the simulator.  (#5629 is the closest issue I see regarding this on quick search).

I think that the correct solution to this is to have SignalLogger set the System version of set_publish_every_time_step to true by default, then allow users to change it's publishing behavior.  I think that each System should be responsible for its default publishing behavior.

sherm1, 2019-01-18:
@edrumwri please close this issue if you agree it is resolved.

closed 2019-01-18 by edrumwri